DeskPi-Team/deskpi #69: Led on power button blink blue/red

Marcone77, 2021-09-07:
Hi Everybody,
my DeskPI has the Led on power button blinking blue/red and the OS (raspianOS) doesn't start.
Any suggestion ?

Thank you Marco

johntdavis84, 2021-09-07:
Red indicates the Desk Pi board is powered off/in standby mode. Blue indicates it’s powered on. 

So it sounds like the board is not getting/keeping power, and is effectively resetting over and over.

Not sure why that would happen. Can you open it up and check all the connections? Maybe try a different power adapter?

pkr1771, 2021-09-07:
Also, check that the larger ribbon cable is correctly attached - I've had power issues when it wasn't perfectly aligned.

Try the board standalone from the case and connections - If it works then the power supply is OK.

johntdavis84, 2021-09-07:
I hadn’t touched a ribbon cable in 20+ years when I assembled this thing.

Everything turned on and stayed on, so I’m assuming it worked. :P What would be the symptoms if it didn’t, aside from infinite resetting.

pkr1771, 2021-09-07:
In my case, the power LED stayed red.  If the Pi is OK when powered alone, try powering when it is connected to the circuit board but without the ribbon cable connected to the Pi.  You can do that without putting inside the case although take care.

Marcone77, 2021-09-07:
Thank you everybody for your replies.
IT seems that the problem is solved....
I've change the power adapter, now I'm using a 45W power supply from my Lenovo Laptop and the RPI4 boot correctly.
I'll do some more testing for few days, but I'm quite sure that we can consider this issue closed.

For you info.....Before to use Lenovo PSU I was using PSU came together with DeskPI.

Thank you everybody

Ciao
Marco

yoyojacky, 2022-03-02:
> 

blinking ?  it will not blinking once connect to power supply, it will still in red.