DeskPi-Team/deskpi #70: Unable to read partition as FAT

NelsonRodMar, 2021-09-08:
Hello,

I post this issue  because I got a error with my Deskpi Pro + Raspberry Pi 4 B 8Go; when I insert my 32Go SD Card formated with Balena Etcher and my OS. When I plugin the SD Card in the Deskpi Pro I got a black screen when I start everything with the following message " Unable to read partition as FAT", message that I don't have when I directly plug this SD Card to my Raspberry Pi.
Why ? Is there a solution ? 

Thank's in advance for you're answer,
Nelson

[image]

yoyojacky, 2021-10-13:
make sure your OS is flashed properly, it seems like your OS is not fit for Raspberry Pi, please download the latest image from www.raspberrypi.org

NelsonRodMar, 2021-11-02:
Thank's @yoyojacky,  that was it.