DeskPi-Team/deskpi #71: Script or Service not working for newest kali-Linux

M3taKn1ght, 2021-09-13:
I'm using the deskpi with two different OS. The first on from SD with Kali-Linux (newest version), the second with the common Raspbian (newest Version)
For both I used the commands in the READ.Me, but it's only working for Raspbian. Kali-Linux has problems with the new services and is unable to start them.
Also the shutdown from kali-linux is no working correct.

M3taKn1ght, 2021-10-18:
Hi yoyojacky, I think this should be implemented in the installation-script for kali-Linux?!

yoyojacky, 2022-03-02:
> Hi yoyojacky, I think this should be implemented in the installation-script for kali-Linux?!

kali 32bit ? or 64bit ? I have tested it on kali and seems like the same result. and i am going to make a selection menu to install script and let user to select the right distribution of OS so that  they may not confuse about the version or distribution.