DeskPi-Team/deskpi #76: [Feature Request] Support for Debian Buster aarch64

F0otsh0T, 2021-10-10:
Requesting support for Debian Buster / aarch64

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

yoyojacky, 2021-11-08:
https://github.com/DeskPi-Team/deskpi/blob/master/install-raspios-64bit.sh  please try this script.

yoyojacky, 2021-11-08:
It has been tested on RaspiOS-64bit-arm-buster and it works fine, please read the readme.md for more details.

yoyojacky, 2021-11-08:
> Impossible for me to install into Raspi Os 64 bits properly
> 
> Errort message like "folder or file not exist" during the install script...
> 
> Waiting for a return of Deskpi Pro Team....

https://github.com/DeskPi-Team/deskpi/blob/master/install-raspios-64bit.sh please try this script on your 64bit OS

closed 2022-06-09 by yoyojacky