DeskPi-Team/deskpi #80: Safe shutdown button issue.

adamsbuilds, 2021-10-19:
Ok, so I have the deskpi driver/software installed.. however the button does not perform a safe shut down...  The button will turn on the pi.. and once on if you long press it will kill power but will not shut down.  Is there some other script that needs installing?  Is there some file that needs to be modified?  I really just want it to work like a safe shutdown script so I push the button and it will close out everything then kill power/shut down.

sondercoder, 2021-10-21:
If there is communication from the MCU to the PI, there should be a method that a short press to run a shutdown script on the PI.  Is there any documentation on the MCU and whats communication method it supports?

adamsbuilds, 2021-10-22:
> @adamsbuilds <https://github.com/adamsbuilds> exactly, that is what I
> assumed when I ordered it.

On a positive note... I came across a Alienware alpha r2.. sff PC. And it
will serve me better for less money.  But yea I do wish the safe shutdown
worked.. may have still hung onto it

yoyojacky, 2021-11-08:
> whats

At recently, MCU only supports following singal which are sending from Raspberry Pi via serial port `/dev/ttyUSB0`:
<pre>
pwm_xxx  - sending pwm_xxx to adjust fan speed, xxx from 0 to 100, 50 means  50% fan speed 
power_off - sending power_off signal to MCU will tell it cut off power after 8 seconds.
</pre>

sondercoder, 2021-11-08:
Can we get the source, or add an option when the MCU senses the button pressed, to send a signal to the Pi, to start a shutdown script?

yoyojacky, 2021-11-08:
> Can we update the MCU firmware (even if it requires JTAG)? It seems rather simple to modify the current shutdown service to not only send a 'power_off' via serial, but to all listen for a message like 'button_request_power_off' and execute 'shutdown -P now' - then as its shutting down, it will send the 'power_off' to the MCU via serial, and a few seconds later, the PI is shutdown, and a few more the MCU powers off the board.
> 
> Is the source code for the MCU, and the build env details available? I would be willing to modify the firmware, and test on my own DeskPi's to add this feature. It would be very useful, and would make the description of the device match the actual capabilities.

Our team has also negotiated for a long time for this. We are afraid that publishing this programming method will cause more problems. If the programming is not good, the entire board will work abnormally and it may cause more trouble. The current deskpi has to go through a lot of tests before it can be released. We have also recorded this function and tried to add it to the later products. The current environment still does not recommend that you burn the firmware yourself. I feel so sorry.

hofnarr132, 2021-12-27:
+1 for this feature (power button press sent to serial port (both long press and short press)) - ideally the power_off delay in the MCU (currently 8 seconds) should be programmable (e.g. pwrdelay_xxx) to allow the current behavior to be overridden and a shutdown script to be run if the OS is still running/operating. Also 'pwrdelay_status' (or similar to return the delay in seconds (e.g. returns 'pwrdelay_008')
It would also be good to get a fan status - e.g. 'pwm_status' (or similar) returns the current value - e.g. 'pwm_050' for 50% fan speed, or 'pwm_000' for fan off.
Is there anything else the MCU could provide? Like it's firmware version?

mysiki, 2021-12-29:
Hello,

your official documentation say : 
```
     Safe-Shutdown & Reset
       After installing a driver, safe shutdown and Reset functions through the onboard MCU are available.
```

I buy this case specially for this feature... And if I understand (maybe it's a leek in my english), this feature is not present ... 

I least, provide minimal tooling and information in order to catch this signal on the pi. 

Otherwise, all this case is really good ! ;)

Superberti, 2022-01-03:
Yes, the announcement on Amazon (here: Germany) is at least misleading. 
On the other side, it should be fairly easy to split the power button wires and to feed it to an raspberry GPIO input. Then a service may watch the power button signal and trigger the shutdown script accordingly. Only thing I don't know is if the MCU is running on 3.3 V or less. Detecting high or low on the Raspi could be a problem then.

yoyojacky, 2022-03-02:
> 

Hi bro, 
you can using pyserial library and wrote a script with python to read information from the /dev/ttyUSB0 device via baudrate 9600,  you will get the some output like "pwm_050" information about the MCU sending out.

yoyojacky, 2022-03-02:
> Hello,
>
> your official documentation say :
>
> ```
>      Safe-Shutdown & Reset
>        After installing a driver, safe shutdown and Reset functions through the onboard MCU are available.
> ```
>
> I buy this [...]

for example : 
```
import serial 

ser = serial.Serial('/dev/ttyUSB0', baudrate=9600, timeout=30)
while True:
     if ser.isOpen():
        data = ser.readline()
        print(data)
```
python script. before you did that ,you may need to install pyserial library by using: 
```
pip3 install pyserial
```

yoyojacky, 2022-03-02:
> 

Hi  dear customer, 
The MCU will reading the signal which sending from Raspberry Pi via `/dev/ttyUSB0` serial device and it will control the fan to spin by the signal `pwm_xxx` xxx is digital number such as `pwm_100` means fan will running at 100% PWM ,  you can just control your program to sending `pwm_xxx` signal to control the fan speed ,and if you sending `power_off` signal, the MCU will cut off the power after 8~15 seconds ..

mysiki, 2022-03-02:
Hi, Thanks for all this information, I try to get the powerOff signal from button but it seems to not be sent to USB.

Using this code : 

```python
import serial, sys
port = '/dev/ttyUSB0'
baudrate = 9600
ser = serial.Serial(port,baudrate,timeout=0.001)
while True:
    data = ser.read(1)
    data+= ser.read(ser.inWaiting())
    sys.stdout.write(data)
    sys.stdout.flush()
```

I only got the fan information 
```
pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025pwm_025
```

But never got any message when I push the power button

yoyojacky, 2022-03-10:
@mysiki , Try to double click the power button, you may get `poweroffpoweroffpoweroff` three times.

Jarod96, 2023-07-27:
Hello guys, sorry for bringing this discussion up again but I am curious if there's any advancement for this missing features! I have the V2, I've updated the SATA board's firmware but still looking for a firmware update for the MCU. 
Any tips will be welcome! 
EDIT: I've tried to requisites a button of the TV's remote controller to be interpreted as a shutdown command but without success at this moment...

yoyojacky, 2024-07-05:
> Hello guys, sorry for bringing this discussion up again but I am curious if there's any advancement for this missing features! I have the V2, I've updated the SATA board's firmware but still looking [...]

Hi jarod96, 
Did you try this :How to Use IR function onboard.
You need to enable gpio-ir function by modify /boot/config.txt file. uncomment this line if not exsit please add it.
dtoverlay=gpio-ir,gpio_pin=17 
Install lirc package:
sudo apt-get install lirc
Modify configuration file on location: /etc/lirc/lirc_options.conf and make sure it has following parameters:
driver          = default
device          = /dev/lirc0
Reboot your Raspberry Pi and test it with following command:
mode2 -d /dev/lirc1 
and then, you can record the TV remoter';s signal and bind to a shutdown script, just like if you read 0x3F, and then execute `sudo sync && sudo init 0` command ? 
and the double click feature is not missing, just because the driver on Raspberry Pi should be installed properly, so that it can grap the signal once you double click the power button, it will trigger shutdown sequnces in a systemd service.

closed 2024-07-20 by yoyojacky