DeskPi-Team/deskpi #86: deskpi.service hangs on start up

w8be, 2021-11-26:
deskpi.service hangs on raspberry pi 64 bit OS.   I can disable the service and manually start the fan but systemctl status shows that the process is Active: starting.   When I manually run the cmd I have to ctr-c to break out.   I am running 

Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

Here is the systemctl status output

`pi@raspberrypi:~/deskpi $ sudo systemctl status deskpi.service
● deskpi.service - DeskPi Fan Control Service
   Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: e
   Active: activating (start) since Thu 2021-11-25 20:59:54 EST; 14s ago
 Main PID: 9185 (pwmFanControl64)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/deskpi.service
           └─9185 /usr/bin/pwmFanControl64 &`

jayc-tech, 2021-12-02:
Hi, I got the same issue. The fan does not start after reboot. I am using the latest raspberry pi 64 bit bullseye.

w8be, 2021-12-03:
I uninstalled the deskpi and reinstalled using:
For Raspbian and RetroPie OS.
cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install.sh
sudo ./install.sh

Since I am running 64 bit  I used the following install cmd:
install-raspios-64bit.sh

I get the same error posted above when the system boots.  I looked at the syslog and see:
Dec  2 22:25:31 localhost systemd[492]: deskpiFan-start.service: Failed at step EXEC spawning /etc/init.d/deskpiFan-start.sh: Exec format error
Dec  2 22:25:32 localhost systemd[1]: deskpiFan-start.service: Control process exited, code=exited, status=203/EXEC

Since the manual control works.  I think I am going to have to write and init.d bash file to run the cmds and bail on the auto start as it does not work on Raspi 64 bit OS.

Would be nice for the developers to chip in and advise but I guess they have moved on.

Wojtasord, 2021-12-17:
Hi, I have the same issue. I am using DietPi64 (under PINN OS). Status shows:

```root@DietPi:/home/rootson# sudo systemctl status deskpi
● deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: activating (start) since Fri 2021-12-17 11:00:01 GMT; 29min ago
   Main PID: 72973 (pwmFanControl64)
      Tasks: 1 (limit: 9354)
        CPU: 496ms
     CGroup: /system.slice/deskpi.service
             └─72973 /usr/bin/pwmFanControl64 & 
```

And it never changes to `Active: active (running)`
In this `activating` state it actually works for about 15 min then fan just stop spinning and does not react to temperature changes. 
For now, I've just  created a simple cronjob with :
``` 
sudo systemctl stop deskpi.service ;
sudo systemctl start deskpi.service &
 ``` 
Every 15 minutes, but this is a silly temp solution, and it would be nice to have it fixed properly. 

@w8be could you share your solution with `init.d` ?

w8be, 2021-12-17:
So my solution with init.d i just start the fan at 50% on boot.  Nothing Fancy.   Got to this [web page](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/) and it shows you 5 different ways to run a progam on boot.   

I created a script called deskpi-init.sh 
I attached it as .txt as I can't attach a .sh file 
put the file in /etc/init.d
make sure the file is a executable 'sudo chmod +x deskpi-init.sh'
then update init.d with  'sudo update-rc.d deskpi-init.sh defaults'

Not elegant but it starts the fan on boot.  
Last item you have to disable deskpi.service in systemctl with 'sudo systemctl disable deskpi.service'

[deskpi-init.txt](https://github.com/DeskPi-Team/deskpi/files/7737817/deskpi-init.txt)

embeddedart, 2022-01-06:
> Hi, I got the same issue. The fan does not start after reboot. I am using the latest raspberry pi 64 bit bullseye.

I also got the sam proble,

hklemp, 2022-01-06:
I had the same problem and adjusted the following in the deskpi.service file 

from "Type=oneshot" to "Type=simple".
from "RemainAfterExit=true" to "RemainAfterExit=no". 

after that the service starts without problems

closed 2024-07-05 by yoyojacky