DeskPi-Team/deskpi #90: [FR] Fedora support

Runkar, 2021-12-25:
Hello!

I switched my homeserver from an x86 host to a raspberry pi4 - because of USB-boot your deskpi case is perfect for this purpose.
But it seems your install-scripts for linux are lacking an option for fedora (or other rpm-based distros) which I prefer.
Maybe there are some more fedora users like me out there who would like to have a convenient way to add your button and fan control.

Cheers!

dustymabe, 2022-05-13:
Fixed with https://github.com/DeskPi-Team/deskpi/pull/82 maybe?

yoyojacky, 2024-07-05:
Hi @Runkar,
 Thanks for your tips. 
 To enable the dwc2 driver on a Raspberry Pi with Fedora OS, you typically need to ensure that the appropriate kernel modules are loaded at system startup and that the firmware is available. Here are some steps based on the search results, which might be applicable to Fedora OS or similar Linux-based operating systems:

Add Kernel Module Loading Instructions: Create or edit a configuration file in the /etc/modules-load.d/ directory (e.g., dwc2.conf) and add the dwc2 module to ensure it is loaded automatically at startup.

Set Up Device Tree: Add or modify configurations in the device tree to enable dwc2 as a USB peripheral. This may involve modifying .dts or .dtsi files and ensuring the correct device tree configuration is loaded at startup.

Install USB Firmware: Make sure the required USB firmware is installed on your system. This can be achieved by adding the linux-firmware package to your image installation list.

Modify Configuration File: Add dtoverlay=dwc2,dr_mode=peripheral to the config.txt file to set dwc2 to peripheral mode.

Install USB Library: Install the libusb1 library, which may be required for interacting with USB devices.

Enable DWC2 Peripheral Mode: Set the environment variable ENABLE_DWC2_PERIPHERAL to 1, which may need to be set when building the system image.

Check Module Loading: Use the lsmod command to check if the dwc2 module is loaded.

Please note that the above steps may need to be adjusted based on your specific system environment and the version of Fedora OS you are using. If you are using Fedora OS, you may need to consult Fedora's specific documentation or community forums for more detailed guidance.

Runkar, 2024-07-05:
Thanks for your advice @yoyojacky.
I will look into it :)

yoyojacky, 2024-07-06:
> Thanks for your advice @yoyojacky. I will look into it :)

 you are welcome ~ , have a nice day~