DeskPi-Team/deskpi #91: VGA monitor

rafealgert, 2021-12-26:
Hi I have a VGA monitor and when i connect to the deskpi pro with the vga to hdmi connector it doesn't display anything. If I connect the same cable to a Pi3 it works. 

I removed the GPIO board just to experiment to see which area is causing the issue. I connected the cable and the monitor showed the login screen. Put back the GPIO board and the monitor indicates no video input, enter sleep mode.

I am using Ubuntu Desktop 64bit 21.10  OS. 

Setting on my /boot/firmware/config.txt 

[pi4]
max_framebuffers=2

[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

# Enable the audio output, I2C and SPI interfaces on the GPIO header
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable the KMS ("full" KMS) graphics overlay, and allocate 128Mb to the GPU
# memory. The full KMS overlay is required for X11 application support under
# wayland
#dtoverlay=vc4-kms-v3d
dtoverlay=vc4-fkms-v3d
gpu_mem=128

# Uncomment the following to enable the Raspberry Pi camera module firmware.
# Be warned that there *may* be incompatibilities with the "full" KMS overlay
#start_x=1

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

hdmi_group=2
hdmi_force_hotplug=1
config_hdmi_boost=4

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
hdmi_drive=2

# If you have a CM4, uncomment the following line to enable the USB2 outputs
# on the IO board (assuming your CM4 is plugged into such a board)

# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2,dr_mode=host

jojobrogess, 2022-03-12:
I think there is a setting somewhere on RPI that allows you to adjust the vga properties of the screen(if screen shows blank, or something like that). If I can find it, I'll post the location

yoyojacky, 2024-07-05:
the possible causes and solutions include:

Adapter Issues: The HDMI to VGA adapter may require a certain amount of current. If the power supply is insufficient, it could lead to video output problems. Ensure you are using an adapter with its own power supply to avoid insufficient power issues.

Resolution Settings: The Raspberry Pi may not correctly detect the resolution of the display, leading to incorrect display. Manually set the hdmi_group and hdmi_mode parameters in the config.txt file to specify the correct resolution.

System Settings: In some cases, you may need to modify the system settings of the Raspberry Pi to adapt to specific display devices. For example, setting hdmi_ignore_edid=0xa5000080 forces the Raspberry Pi to ignore the EDID information of the display and output at the specified resolution.

System Compatibility: The list of operating systems supported by DeskPi Pro includes some that may no longer be supported or have compatibility issues. Ensure that the operating system you are using is a version that has been tested and supported.

Firmware Configuration: You may need to modify the firmware configuration file of the Raspberry Pi, such as /boot/firmware/config.txt, to enable or configure specific features, such as power management for USB ports.

Hardware Compatibility: DeskPi Pro is designed for the Raspberry Pi 4, ensure that the adapter and display are compatible with the HDMI output of the Raspberry Pi 4.

Software Updates: Make sure that the software and firmware of DeskPi Pro are up to date to get the best hardware support and compatibility.

If you encounter display issues, try the above solutions and adjust according to the specific situation.

closed 2024-07-05 by yoyojacky