DeskPi-Team/deskpi #94: No SSD

Swifster, 2021-12-29:
new Pidesk installation new RPI4 tested with both Buster and Bullseye

Did a update of Deskpi firmware and nothing no disk shows up. Now raw disk, no formatted disk, no disk with image, nothing.
[   78.255076] usb 1-1.1: device descriptor read/64, error -110
[   80.164710] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?

No disk shows up when i plug the usb adapter or us a extra cable. It only shows up when i plug it to the lower USB 3 on the raspberry.
usb 2-2: new SuperSpeed Gen 1 USB device number 9 using xhci_hcd
[  487.147990] usb 2-2: New USB device found, idVendor=174c, idProduct=55aa, bcdDevice= 1.00
[  487.148010] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[  487.148027] usb 2-2: Product: Pro
[  487.148054] usb 2-2: Manufacturer: DeskPi
[  487.148068] usb 2-2: SerialNumber: 002020110210

jojobrogess, 2022-03-16:
Ermmmm. Try to see if you can update the daughter board using the supplied zip. That's my best suggestion

Swifster, 2022-10-10:
Hi

Problem was the PI there was issue with the usb 3 ports. I went over the solder point’s with hot iron and afterwards al was working.

closed 2024-07-05 by yoyojacky