DeskPi-Team/deskpi #95: Power Button Doesn't Work

YourFriendCaspian, 2022-01-21:
The button on the front doesn't do anything when pressed or held. The light is on and blue but never changes to red no matter what state the pi is in. I ran the install script, I set the fan with the deskpi-config script so I know it installed correctly, the front usbs work but the button does nothing. I'm on the legacy version of Raspberry Pi OS by the way. Do I just have a defective daughterboard or is there something I'm missing? It already was disappointing that it doesn't have proper safe shutdown like it advertised but to just have an always on blue light on the front is just ridiculous.

Aultruist, 2022-01-26:
I'm having the same problem. Just installed Deskpi Lite - the power button did turn off initially when shutting down my RP4 running RetroPi, but now it just stays on (blue), and I have to cut the power manually. I am noticing the following failure statement in the shutdown script: "Failed to start DeskPi Safeshutdown Service".
I'm also having problems getting Deskpi-config to control fan speed. I can get Raspi-config to control the fan (I turn the fan on at 60 degrees), but the fan speed is either 100% or off. Some troubleshooting documentation would be helpful.

YourFriendCaspian, 2022-02-01:
I switched to the most recent version of Twister OS and am having the same problem with the button. It does nothing but the blue light. Pressing or press/holding the button does absolutely nothing.
I'm also having the same problem that others are reporting with the fan control not turning on the fan at set temperatures. If I rerun the deskpi-config script the fan will run for a bit but almost any normal use will make it stop again.
Do I just have a bad daughterboard? If there's nothing I can do because it's probably a bad board, can I get a replacement? Please advice.

Exarkuniv, 2022-02-14:
@YourFriendCaspian no its not you, i just got my deskpi lite and i also cant get anything to work, its a nicer case then what i had before, but the support here is very lacking.

im just going to have to mod it to make it work the way i need it to with RetroPie

kiraakischona, 2022-02-18:
same here with deskpi lite, using libreelec/Kodi. If anyone could help, pls...

Exarkuniv, 2022-02-22:
so after messing with this, it seems the script does nothing. the power on and off is hardwired in to the circuits of the daughter board we plug into the Pi. 
you can see the power switch go from red to blue and back again without the Pi plugged in. 
it will always just cut the power. 

so me i just found out what to pull off to stop the button from powering off the daughter board and then rewired it to make it have a safeshutdown. 
sad that they did not do it right. i know that other companys out there have done it just fine. 
oh well i still like the case and all

zen0bit, 2022-02-22:
> so after messing with this, it seems the script does nothing. the power on and off is hardwired in to the circuits of the daughter board we plug into the Pi. you can see the power switch go from red [...]

I am curious what to do..

Exarkuniv, 2022-02-22:
@zen0bit 
how comfortable are you with soldering?

kiraakischona, 2022-02-22:
Looking forward for your explaination about what to do, but for now, thanks for the tests and the time you've invested. I do like the case aswell, but it would be great to use it as supposed (or intended)...

Exarkuniv, 2022-02-22:
i will link pictures of what i did and do my best to explain it 

it does involve soldering, not much if you just want the power button to work, if you want the LED in the power button to work that does take a little more

Exarkuniv, 2022-02-24:
DISCLAMER. I take no responsibly for any damage done. (i already damaged mine doing this)

1. This is the pin the controller chip uses to detect the button push. 
2. This is there the button pin out goes to first before to the chip.
3. This is there the resistor is for the red LED of the button.
4. This is there the resistor is for the blue LED of the button.
5. This is there the resistor is for the blue LED of the button.
Also i removed all the GPIO pins on top of the fan/heatsink since i will not be using them

First i have the switch set for all ways on. 

i did unsolder the 1. pin from the board, but i dont thing that is needed. i cant test it with it connected again since the pad got messed up when i removed it.

next i connected a wire to 2. that is going to the GPIO pin that my shutdown script. you will connect it to wherever the pinout for what you use tells you. 

next i removed the resistor on 3, 4, 5 as these are for the LED. I dont know what 5 did but it did power on the LED so i took it away. Next i reused the resistor for 4 again and connected it to the UART pin.

i was trying to see if i could figure out how to get the red LED to light when the Pi is off, using a transistor but i blew the red LED part so i cant do much with that now.

well thats it. i hope this helps someone

[image]
[image]

Terumasa1972, 2022-07-21:
I hope connecting the switch and gpio3 would go well.

yoyojacky, 2024-07-05:
> The button on the front doesn't do anything when pressed or held. The light is on and blue but never changes to red no matter what state the pi is in. I ran the install script, I set the fan with [...]

If you want to turn off the power button light,( from blue to red), your system needs to send `POWEROFF` from Raspberry Pi to the  `/dev/ttyUSB0`  which is MCU's serial port on deskpi daughter board, and the mcu will detect the `POWEROFF` signal and then waiting for about 8-10 seconds to cut off the power, and then, the led on power button will turn to red, it means the daughter board has cutted off the power.  and at the same time, if you double click the push button, it will sending `poweroff` 3 times from MCU to Raspberry Pi via `/dev/ttyUSB0` port, it means that you can trigger safe shutdown sequnces from raspberry pi once your serial port got this `poweroff` signal. 

For short: 
if: 
Raspberry Pi  got signal `poweroff`  <--------  double click power button  ,  you can initialize the shutdown procedure via program, or python scripts.
if:
MCU on deskpi daughter board got signal `POWEROFF`  <-------------- Raspberry Pi OS sending `poweroff`  via `/dev/ttyUSB0` 
MCU will cut off the power, turn red the led indicator on Power button.

closed 2024-07-05 by yoyojacky