DeskPi-Team/deskpi #96: Fan not running on Raspbian BullsEye 32bit

bjs-pdx, 2022-01-23:
Recently installed my Pi4 into the DeskPi Pro case.  Unable to get the fan to spin-up at all even if set to run at 100% all the time.

Some information:

pi@pi9:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-01-23 14:56:25 PST; 2s ago
   Main PID: 1328 (sudo)
      Tasks: 2 (limit: 4915)
        CPU: 38ms
     CGroup: /system.slice/deskpi.service
             ├─1328 sudo /usr/bin/pwmFanControl
             └─1329 /usr/bin/pwmFanControl

Jan 23 14:56:25 pi9 systemd[1]: Started DeskPi PWM Control Fan Service.
Jan 23 14:56:25 pi9 sudo[1328]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
Jan 23 14:56:25 pi9 sudo[1328]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)

bjs-pdx, 2022-01-24:
Did a clean install of Raspbian Buster on SD Card and fan is still not functioning correctly.  Set four points as 100% speed and no fan.  Set default speed to 100% and no fan.

Contents of /etc/deskpi.conf:

0
100
50
100
65
100
75
100

Last lines of /boot/config.txt

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=dwc2,dr_mode=host

Am I missing something needed to get this working?

bjs-pdx, 2022-01-25:
Trying to troubleshoot this whole thing.  Where does deskpi-config log operations and changes?  When I change the setting from manual set points to option 4 (100%) why do I still have the /etc/deskpi.conf file with the original settings?  Should the changes appear in syslog?  Should I have the directory /home/pi/deskpi?  I don't...

bjs-pdx, 2022-01-27:
So, today the fan is working.  If I run deskpi-config without sudo and select a speed of 100% the fan turns on and continues to run, as long as I don't reboot.  Seems if I reboot the pi the settings go back to default and the fan only runs once those temperatures are reached.  Is there some way to set the fan to always run at 100%?

bjs-pdx, 2022-02-10:
Noticed today the Pi was getting hot and discovered the fan was not running after a reboot.  Used deskpi-config to set the fan to 100% but it should have been doing that already since I set the temperatures in /etc/deskpi.conf to run the fan on high speed all the time...what gives?

contents of /etc/deskpi.conf
5
100
10
100
15
100
20
100

D-A-H-8-5, 2022-10-01:
Did you ever figure this out? I am having the same issue?

yoyojacky, 2024-07-05:
please download the preinstalled OS : https://drive.google.com/file/d/1fKJ6ZpMaEAjIvEa72R3Y0AAQbiwLF76c/view?usp=drive_link 
and flash it to your TF card, do not need to install any driver, just power up. 
[image]

closed 2024-07-05 by yoyojacky