DeskPi-Team/deskpi #97: Auto Power-On after a Power Failure

ksdhans, 2022-01-27:
I'm using the DeskPi Pro as a server, and I'd like it to switch itself back on automatically after a power failure. This is especially important if I'm away (e.g., working remotely), so I can't just walk over and push the power button. 

Is there a way to configure the hardware to auto-power on after a power failure? Could this be added?

I'd prefer a solution that doesn't involve modifying the hardware if possible.

paulpeeters, 2022-02-16:
I'm also interested in such a feature similar to how this can be set in the bios of a PC.

ksdhans, 2022-02-17:
The "safe power off" script sends the string "power_off" over the serial port from the Raspberry Pi to some device on the DeskPi pro's circuit-board. I wonder if that device has other commands that could configure the power button behaviour.

We really need someone from the DeskPi-Team to respond...

Superberti, 2022-03-02:
I would be interested in a power-on-after-reboot solution, too. For me, this is more important than the missing safe-shutdown-on-button-press feature.
Nice case, but still in some kind of beta status...

jojobrogess, 2022-03-16:
I looked into this, I wasn't able to find a fix. 

Even with setting the power button to always on and messing with the code.

I had to resort to getting a Wi-Fi connected power hub, to "flip the switch" when the power would come back on.

ksdhans, 2022-03-18:
@jojobrogess 
That's disappointing. How does your WiFi connected power hub "flip the switch?"

ksdhans, 2022-03-22:
This is going to make everyone feel silly. Page 9 of the DeskPi Pro manual shows a toggle-switch that you can flick to enable "always on" mode. You can get at it by unscrewing the front panel.

[image]

closed 2022-03-22 by ksdhans

jojobrogess, 2022-03-22:
It's on the original as well.
I have a v1.

I'm pretty sure I have my case set to always on mode and after a power outage it still stays off.

BentBanana, 2022-08-29:
I have the new v2 and it doesn’t work. Has a work around been identified?

wpietri, 2023-03-05:
In case anybody ends up here looking for this: For the DeskPi Lite, there is a switch near the power button labeled "D - E". Putting it in the E setting enables auto power on.

BentBanana, 2024-06-02:
I had the same issue on the DeskPi V2. What I did was put a screw driver on top of the switch housing that selects auto reboot as in the pictures above and pushed it down. It clicked into place and then worked.

zazafree00, 2024-07-15:
> In case anybody ends up here looking for this: For the DeskPi Lite, there is a switch near the power button labeled "D - E". Putting it in the E setting enables auto power on.

Thank you, this worked for me perfectly.