DeskPi-Team/deskpi #102: Does the fan speed control systemd script work with DeskPi Lite?

CaptainMidnight, 2022-02-21:
With DeskPi Lite, the inbuilt PiOS pwm fan control appears to work i.e. off until temp reaches 60c.

Installing deskpi-config, manually changing the fan speed doesn't appear to work.

Using deskpi-config to turn fan on/off doesn't appear to work.

If this script is compatible with DeskPi Lite what combination of settings are required to enable a working deskpi-config utility - does the PiOS pwm fan control need to be enabled/disabled for it to function properly?

What diagnostics can I run to debug and find out what's going wrong - ideally I'd like to use option 6 and have set fan speeds like the default between 40c to 60c (i.e. no fan, 25, 50, 75 then 100%).

Any help with this would be greatly appreciated.

Hardware: Pi4B8, DeskPi Lite
Software: PiOS 64-bit, deskpi-config

CaptainMidnight, 2022-03-05:
The PWM element of the fan is fully controllable, you just need the right coding method: -

[Link to 52PI PWM fan ZP-0110 configuration example](https://wiki.52pi.com/index.php?title=ZP-0110)

CaptainMidnight, 2022-03-06:
@yoyojacky after some testing and review of the DeskPi Lite daughter board circuitry, the fan control scripts provided by this github need to be rewritten with respect to the DeskPi Lite case as it has the following issues: -

- IT IS IMPOSSIBLE TO CONTROL the PWM fan using the current /dev/ttyUSB0, the actual PWM fan connection inside the Lite version of the case is DIRECTLY CONNECTED TO THE Pi4 GPIO header, therefore sending instructions i.e. fan speed control signals like 'pwm_100' have no way of the daughter board MCU changing the GPIO pin on the Pi4 header as it has no logical connection to make those changes.

- The current DeskPi PWM fan control scripts/driver needs to be rewriiten to provide a separate DeskPi Lite version of the install software - this 'Lite' version needs to use and make standard GPIO PWM value changes NOT SENDING COMMANDS to /dev/ttyUSB0, as this has NO FUNCTIONALITY on the "DeskPi Lite" due to how the PWM fan is hardwaired to the Pi GPIO header.

- As already stated else where, the daughter board MCU has no ability to allow for safe shutdown of the Pi4 by pressing the front power button as the MCU sends no constant 'power_on' stafus back to the Pi4 - although regrettable, this limitation of the current daughter board MCU programming is accepted and understood. 

The DeskPi Lite is a great case with known limitations i.e. the power button, but if the PWM fan control script was rewriiten to 'Lite' specific i.e. to use GPIO PWN control, this would help to make this case truely way better than it currently is as received by customers trying to use the supplied INCOMPATIBLE PWM fan control service.

@yoyojacky can you confirm if the MCU PWM fan control functionality is really for the 'empty' daughter board fan socket - this would make sense why in the 'Lite' implemented hardware it doesn't do anything?

CaptainMidnight, 2022-03-06:
@yoyojacky PWM fan control scripts need to be rewritten and offered via a separate dedicated 'DeskPi Lite' install script to reflect that the 'Lite' case hardware requires use of GPIO PWM fan control commands and not '/dev/ttyUSB0' fan control commands.

CaptainMidnight, 2022-03-06:
https://github.com/DeskPi-Team/deskpi_v1/issues/2#issue-1160561578

yoyojacky, 2022-03-10:
@CaptainMidnight yes, it has been fixed. thanks for your tips..

closed 2022-03-10 by yoyojacky