DeskPi-Team/deskpi #103: Is anyone actually maintining this software?

bjs-pdx, 2022-02-22:
There are multiple posts regarding problems with the software but only crickets from the dev.  Is anyone actually maintaining this code and repository or is it just dead?  I'd really like to know since it seems nothing is getting fixed.

jojobrogess, 2022-03-16:
It doesn't look like it, tbh.

yoyojacky, 2023-02-17:
> There are multiple posts regarding problems with the software but only crickets from the dev. Is anyone actually maintaining this code and repository or is it just dead? I'd really like to know since it seems nothing is getting fixed.

I will follow up the case and maintain this repository .

bjs-pdx, 2023-06-12:
So, looking at the dates on the code files, this still isn't being updated or maintained.  What a pity.  Really great case for the PI but really poor support.

yoyojacky, 2024-05-13:
> So, looking at the dates on the code files, this still isn't being updated or maintained. What a pity. Really great case for the PI but really poor support.

sorry for my slow response. I have pushed a new brench to support bookworm right now.

closed 2024-05-13 by yoyojacky