DeskPi-Team/deskpi #105: DeskPi Pro Fan Setting

JCB2277, 2022-03-13:
deskpi-config works to set the fan speed to option 4 - 100% but it drops out after a short period of time and the fan seems to work but not at the 100% setting, Also if you reboot the Pi you must run deskpi-config again to set the fan speed. I have a Pi 4b with 4 gig of ram booting from a 512gb SSD overclocked to 2000

jojobrogess, 2022-03-16:
OS system?

bshimmy, 2022-05-13:
I have the same issue.

RPI 4b, 8GB RAM, using SSD for storage, running Raspberry Pi OS (64-bit).

administrator@rpi4:/etc $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

BertMedley, 2022-09-18:
Status?  I just bought a deskpi pro.

JCB2277, 2022-09-18:
You still have to set the fan speed each time you reboot your Pi - I have 5 of the DespkPi Pro units and love them other than the issue with fan speed

yoyojacky, 2023-02-17:
OK, everybody, I am going to rewrite te installation script and the fan control script for you guys. sorry for bring you so much confuse ...

closed 2024-07-05 by yoyojacky