DeskPi-Team/deskpi #106: deskpi lite config file location

crusader27529, 2022-03-13:
I'd like to look at or edit the config file, but I've no idea where it is........or what it's called.

crusader27529, 2022-03-13:
The request was to possibly change a timer value that another thread mentioned to fix the power button issue. The config.txt file is an executable script with no actual parameters to change.

Maybe a schematic of the daughter card would help so we could make the button actually do something useful. It does nothing now.

I bought this case for 2 reasons, with the primary was to be able to do a safe power shutdown. The cooling was secondary.

I have othe RPI with external PB's that cause the unit to go to a low power state after doing a shutdown, and then starting up when the button is pressed again. That functionality would be an adequate alternative function. I don't really think that the deskpi lite is actually capable to switch the power off. It probably also goes into a low power state anyway.

BTW, my other RPI autostarts after a power failure, also.

jojobrogess, 2022-03-14:
Need to know what OS you're installing to, to know where it has been installed.

crusader27529, 2022-03-14:
Raspios 64 bit

jojobrogess, 2022-03-14:
from the installer `install-raspios-64bit.sh`

*******************************************************************

/usr/bin/pwmFanControl64
/usr/bin/safecutoffpower64
/usr/bin/deskpi-config

/boot/config.txt

fanDaemon="/lib/systemd/system/deskpi.service"
pwrCutOffDaemon="/lib/systemd/system/deskpi-cut-off-power.service"

closed 2024-07-05 by yoyojacky