DeskPi-Team/deskpi #107: DeskPi Pro V2 Script

djc1961, 2022-03-16:
I have a Raspberry Pi 4 running the standard Raspberry Pi OS Lite (no GUI) and OpenMediaVault 6.0. The Pi is fitted inside the DeskPi Pro V2 case along with 1 TB M.2 SATA memory.

The assembly of the case went Ok and I was able to start and run the Pi and OMV without any problem. I was also able to access the M.2 memory

When I tried to install the script for the DeskPi Pro V2 I ran into trouble

My OS, the standard Raspberry Pi OS Lite, does not appear to be specifically listed so I tried the script for the standard Raspberry Pi OS full (with GUI).

The script I tries is that below.

cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install.sh
sudo ./install.sh

When I try to execute the second line (git clone ......) the command is not recognised.

Can you pleas offer a solution?

Many thanks for the great product.

jojobrogess, 2022-03-16:
AFAIK, there is NO V2 script. 

All scripts should work for both cases(Pro/Lite) equally. 
And by equally, I mean terribly. 

********************* 

Git clone is a weird way to download from the internet. 

If git clone doesn't work in your terminal, then you don't have git installed.?

********************* 

Might be better to:

wget URL2zip 
unzip zipFile

Then:

Chmod +x zipDirectory/install.sh 
./install.sh

********************* 

But if you don't know how to do that yourself, this case was probably a bad purchase. 

As most operating system support is limited at best, and/or functionally sporadic. 

I would classify this case, as a “makerboard”. Considering limited support. 

********************* 

You might have to customize the install script to work correctly for your OS. 

GOOD LUCK

jojobrogess, 2022-03-17:
sorry i misread that you had a lite, you have a pro v2. 
same solution though.
if git clone doesn't work. 

********************************************************************

`wget https://github.com/DeskPi-Team/deskpi/archive/refs/heads/master.zip`
`unzip deskpi.zip `    
(i think that's what it saves as. you can double check with a simple `ls` command and replace it if it's different)

Then:

`Chmod +x deskpi/install.sh`
`./install.sh`

djc1961, 2022-03-17:
Many thanks for the advice. I will give it a go when I get time, 
probably not until Saturday

David

closed 2024-07-05 by yoyojacky