DeskPi-Team/deskpi #109: Cannot install latest 64bit RaspiOS on DeskPi Pro v2

Taomyn, 2022-04-02:
Just spent half a day trying to get the latest version of 64bit RaspiOS (2022-01-28) on my DeskPi Pro v2, 4Gb RPi4, and it just hangs, both from SD card and USB/SSD - it boots, I see the console messages, then the screen blanks and nothing more. The LEDs on the RPi go dark as well. There is no response nor is there any network connection although the LAN adapter lights remain on and solid.

I tried the image as per the documents, an image from May 2021 so not something I can use live, but that works fine. I was able to get connected via the LAN and check that the firmware of the Pi was up-to-date, but that's as far as I want to go with that version of the OS.

Any ideas?

jojobrogess, 2022-04-11:
Issue looks like  [#110](https://github.com/DeskPi-Team/deskpi/issues/110)

More than likely it was because of the boot process(modifying config.txt), or the systemctl stuff.

You need to look into what happened/how to fix it all from outside of the OS. 
Once you can boot in, look at what the installer installs(or sys logs to see where it stopped(if it stopped)) and delete the files it installed. 
More than likely from there, you'll understand why it went bad in the first place and you can fix the rpi 64bit installer.

or someone can offer a solution. 
BUT to get it to run again, without loosing data, you need to externally load the drive as a secondary drive and delete the things the installer messed with.

Taomyn, 2022-04-12:
Sounds familiar because that's what happens between trying 64bit and then reverting back to the previous drive - I've been putting the new OS on a different SSD to test. To fix that I had to boot once from just the SD card with a fresh OS 32bit or 64bit, it does not matter, then the next boot with just the SSD the Pi is fine again.

It was also not possible to boot in and look at anything as you suggest - the RPi was basically dead until switched off and back on.

Some good news, it seems the recent update they released last week does work fine now and I was able to start rebuilding the whole machine. Unfortunately I had to revert again because one service I run is not compatible yet with 64bit.

To me, whatever the problem was before is now resolved with the new version of RaspiOS - hope in the coming weeks/months I'll be able to try again.

closed 2022-04-12 by Taomyn