DeskPi-Team/deskpi #110: Pi 4B is now bricked.

RobertKing42, 2022-04-09:
I installed the latest 64-bit Raspberry on an SSD connected to a Pi4.  All went well and I was able to make a lot of progress getting everything set up.

This I installed the DeskPi software:

cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install.sh
sudo ./install.sh

Now the system is bricked.  I get the Pi splash screen and then nothing except a solid SDD activity light.

jojobrogess, 2022-04-11:
I'm pretty sure bricked implies you CANNOT get your device to run again, ever, and I would assume if you re-flash your SSD it would still work, or if you used a different drivemedium it would boot up fine. 

You'd loose all your data though, obviously. You might be able to load the drive as a secondary drive to another drive, and grab files out that way. 

**********************************************************************

I've ran through their installer for the rpi quite a bit, I don't think there was anything in there that would ever do something like this. The only things I can think of are the config.txt code(to turn on the serial, usbboot, and ir) and possibly the systemctl stuff. But even then, not loading in again? Seems fishy. More than likely you had something running/configured that conflicted with the install process.

**********************************************************************

Plus you don't really give info on how it happened:

Did you try to install, and midway through it shutoff? Or After? 
Or was it after you rebooted? Or did you shutoff and turn back on and then it didn't boot in?

**********************************************************************

You can load the drive as a secondary drive and pull up system logs to see what happened.

Also try a different HD, like I said above, if it worked. The boot process for your RPI 64bit SSD is burnt, and you need to look into what happened/how to fix it all from outside of the OS.  Once you can boot in, look at what the installer installs(or sys logs to see where it stopped(if it stopped)) and delete the files it installed. More than likely from there, you'll understand why it went bad in the first place and you can fix the rpi 64bit installer. 

Or maybe someone else has experienced this and can help out. 
idk what happened/or how to fix without more explaination/sys logs.

dustymabe, 2022-06-03:
> Now the system is bricked. I get the Pi splash screen and then nothing except a solid SDD activity light.

I got my deskpi a few weeks ago and have tinkered with it a bit. What I've noticed is that my deskpi can get into the situation you describe, but I haven't installed the software from this repo yet at all. I notice there is a sweet spot for the USB A-A adapter thing where if you have it pushed in too far the system won't see the USB HDD at all. If you pull it out ever so slightly it will then work.

This is my experience. Hope it helps you too.

johntdavis84, 2022-06-03:
You might also want to try swapping out the USB-C power adapter for a different one with equal or greater amps.

The one that came with my Desk Pi started making a horrifying buzzing noise about 4 hours after I plugged it in. The sound of a capacitor about to fail. Yours could be silently misbehaving.

If you’re in the US, a 5V/3A USB-C cell phone wall charger should be fine.

yoyojacky, 2024-07-05:
installation script will not effect to your system..... did you get any message on screen ?

closed 2024-07-20 by yoyojacky