DeskPi-Team/deskpi #111: Script for Armbian x64

toprisan, 2022-04-12:
Is there a script that works for armbian 64 bit? Apparently the ubuntu one doesn't cut it. Or can you provide some pointers?
I'm sitting on 2 units now, and I can't use the features of the enclosure until I have the script installed.

jojobrogess, 2022-04-13:
Look at installer, find out what is going on, look for similar file locations in your OS. 

Change installer to use those locations. 

I doubt you'll get one made for you, I just spent a couple weeks(maybe a month) teaching myself shell, bash, and python in order to get mine installed on Libreelec.

yoyojacky, 2024-05-13:
I just put the brench to this repository,  feature/bookworm, hope it can help you work it out.

closed 2024-07-05 by yoyojacky