DeskPi-Team/deskpi #113: Missing ttyUSB0 in Ubuntu Jammy

mazzystr, 2022-05-05:
For Ubuntu Jammy If you notice that your /dev/ttyUSB0 device is missing it could be because another character device has taken control of the ttyUSB0 and renamed it.

Check dmesg

Then check to see if you have braille tablet installed.  Uninstall the pkgs if so
`apt remove brltty libbrlapi0.8  python3-brlapi xbrlapi`

m0dded, 2022-05-16:
Hi - i still can't get the ttyUSB0 to work. I do not have any of those apt packages.
Hoping you can help!

My lsusb output
```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

mazzystr, 2022-05-16:
Can you post output from `dmesg | grep tty`

m0dded, 2022-05-16:
Hi 
Sure - thanks for helping!

```
pi@raspberrypi:/opt$ sudo dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:CD:3F:7F vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyS0,115200 dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 rootwait fixrtc quiet splash
[    0.000327] printk: console [tty1] enabled
[    1.066279] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 17, base_baud = 0) is a PL011 rev2
[    1.075057] printk: console [ttyS0] disabled
[    1.075114] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 19, base_baud = 62500000) is a 16550
[    1.083080] printk: console [ttyS0] enabled
[    6.711668] systemd[1]: Created slice Slice /system/serial-getty.
[    7.876161] usb 1-1.2: cp210x converter now attached to ttyUSB0
```

Output of uname -r
```
pi@raspberrypi:/opt$ uname -r
5.15.0-1006-raspi
```

Output for dwc2
```
pi@raspberrypi:/opt$ sudo dmesg | grep dwc2
[    1.803763] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    1.803923] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    1.859247] dwc2 fe980000.usb: DWC OTG Controller
[    1.859388] dwc2 fe980000.usb: new USB bus registered, assigned bus number 3
[    1.859430] dwc2 fe980000.usb: irq 21, io mem 0xfe980000
[    1.859636] usb usb3: Manufacturer: Linux 5.15.0-1006-raspi dwc2_hsotg
```

Works in Raspbian.

Output of uname -r
```
5.15.32-v7l+
```

Output in Raspbian for dwc2
```
pi@raspberrypi:~ $ sudo dmesg | grep dwc2
[ 6.571855] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[ 6.623458] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[ 6.745506] dwc2 fe980000.usb: DWC OTG Controller
[ 6.745560] dwc2 fe980000.usb: new USB bus registered, assigned bus number 3
[ 6.745624] dwc2 fe980000.usb: irq 36, io mem 0xfe980000
[ 6.747266] usb usb3: Manufacturer: Linux 5.15.32-v7l+ dwc2_hsotg
[ 7.182279] usb 3-1: new high-speed USB device number 2 using dwc2
[ 7.872295] usb 3-1.4: new full-speed USB device number 3 using dwc2
```

Can you please give me lines from your config.txt and cmdline.txt ?
Are these files in /boot/firmware or in /boot ?

mazzystr, 2022-05-16:
I wanna focus on your dmesg output
```
[    7.876161] usb 1-1.2: cp210x converter now attached to ttyUSB0
```
That is not correct.

If all was good it will look like
```
[   69.786658] usb 3-1.4: ch341-uart converter now attached to ttyUSB0
```

Verify your cmdline.txt file.  Mine looks like the following.  I am booting from sdcard.  I don't have a min to boot from crypt/lvm just yet.
```
# cat /boot/firmware/cmdline.txt 
zswap.enabled=1 zswap.zpool=z3fold zswap.compressor=zstd dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 rootwait fixrtc quiet splash
```

Verify your /boot/firmeware/config.txt.  Mine can probably use tuning but it works.
```
[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

[pi4]
max_framebuffers=2
arm_boost=1

[all]
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on
disable_overscan=1

[cm4]

[all]
dtoverlay=vc4-kms-v3d
camera_auto_detect=1
display_auto_detect=1
arm_64bit=1
dtoverlay=dwc2,dr_mode=host
over_voltage=6
arm_freq=2000
gpu_freq=750
```

m0dded, 2022-05-17:
> I wanna focus on your dmesg output
> 
> ```
> [    7.876161] usb 1-1.2: cp210x converter now attached to ttyUSB0
> ```
> 
> That is not correct.

This is my sonoff zigbee dongle - so in raspbian it just moves automatically to ttyUSB1 - but because my ch341 is not being recognized Ubuntu is allocating ttyUSB1 to it.

> If all was good it will look like
>
> ```
> [   69.786658] usb 3-1.4: ch341-uart converter now attached to ttyUSB0
> ```
>
> Verify your cmdline.txt file. Mine looks like the following. I am booting from [...]

Mine looks like this. I am using M.2 Drive
```
pi@raspberrypi:/opt$ cat /boot/firmware/cmdline.txt 
console=serial0,115200 dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 rootwait fixrtc quiet splash
```

> Verify your /boot/firmeware/config.txt. Mine can probably use tuning but it works.
>
> ```
> [all]
> kernel=vmlinuz
> cmdline=cmdline.txt
> initramfs initrd.img followkernel
>
> [pi4]
> max_framebuffers=2 [...]

This is mine:
```
pi@raspberrypi:/opt$ cat /boot/firmware/config.txt 
[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

[pi4]
max_framebuffers=2
arm_boost=1

[all]
# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
# parameters related to the base device-tree they must appear *before* any
# other dtoverlay= specification
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# Enable the serial pins
enable_uart=1

# Autoload overlays for any recognized cameras or displays that are attached
# to the CSI/DSI ports. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=1
display_auto_detect=1

# Config settings specific to arm64
arm_64bit=1

[cm4]
# Enable the USB2 outputs on the IO board (assuming your CM4 is plugged into
# such a board)

[all]
dtoverlay=dwc2,dr_mode=host
```
The only difference I see is mine has enable_uart=1 and yours doesn't ?

What is your output of uname -r ?

m0dded, 2022-05-17:
ok so strange. All along I was doing a soft reboot. It just occurred to me that since the ports are part of the USB-C (power port) - I unplugged the power cable. And plugged it back in and now I can see the ports!

Thanks for assisting.

iainaff, 2022-06-15:
Check my patch in issue #118 to use a dynamically named device, rather than a hard-coded /dev/ttyUSB0 (which is very bad practice, anyway). 

It may help.

https://github.com/DeskPi-Team/deskpi/issues/118