DeskPi-Team/deskpi #116: Fan Error Ubuntu Server 64bits 20.04

josemmunoz, 2022-06-04:
`ubuntu@control01:~/deskpi$ sudo ./install-ubuntu-mate.sh
 * DeskPi Fan control script installation Start.
 * DeskPi main control service loaded.
cp: cannot stat '/home/ubuntu/deskpi/drivers/c/pwmFanControl': No such file or directory
chmod: cannot access '/usr/bin/pwmFanControl': No such file or directory
 * DeskPi Service configuration finished.
 * DeskPi Service Load module.
Created symlink /etc/systemd/system/multi-user.target.wants/deskpi.service → /lib/systemd/system/deskpi.service.
Created symlink /etc/systemd/system/halt.target.wants/deskpi-safeshut.service → /lib/systemd/system/deskpi-safeshut.service.
Created symlink /etc/systemd/system/shutdown.target.wants/deskpi-safeshut.service → /lib/systemd/system/deskpi-safeshut.service.
Created symlink /etc/systemd/system/poweroff.target.wants/deskpi-safeshut.service → /lib/systemd/system/deskpi-safeshut.service.
 * DeskPi PWM Fan Control and Safeshut Service installed successfully.
 * System will reboot in 5 seconds to take effect.
Job for deskpi.service failed because the control process exited with error code.
See "systemctl status deskpi.service" and "journalctl -xe" for details.
^[[2~Reboot system for changes to take effect`

I have too much issues whit ubuntu 20.04 server

jojobrogess, 2022-06-11:
Make sure the Deskpi downloaded files are at that location or change location in installer file to where you do have them, then rerun.

jonhadfield, 2022-07-21:
This PR fixes the issue with the `cp` command: https://github.com/DeskPi-Team/deskpi/pull/125. The reason is that compilation of that binary fails due to missing dependencies that this PR addresses.

There's another PR, for the same script, to suppress another error you probably faced: on first run, it tries to backup a file but fails: https://github.com/DeskPi-Team/deskpi/pull/126.

jzazo, 2022-07-24:
When will these two be merged? Also, it seems there is no uninstall script for ubuntu 64, can it be added? Thank you!

yoyojacky, 2023-02-17:
> "'/home/ubuntu/deskpi/drivers/c/pwmFanControl'"

sorry, i will fix it at once..

yoyojacky, 2023-02-17:
> When will these two be merged? Also, it seems there is no uninstall script for ubuntu 64, can it be added? Thank you!

OK,  I am working on it.

closed 2024-07-05 by yoyojacky