DeskPi-Team/deskpi #122: Home Assistant HAOS Support

sgreenslade, 2022-06-27:
Hi,
Does anyone have this installed and working with Home Assistant OS?

There are some instructions on how to access the underlying OS here: https://developers.home-assistant.io/docs/operating-system/debugging/
But will the shutdown scrip gracefully stop the Docker Containers?

al-jimenez, 2022-07-04:
I have also been looking for this...   

I have found this, but have not tried it yet..... https://community.home-assistant.io/t/deskpi-pro-active-cooling-and-front-usb-ports/265136

andymaycoates, 2023-01-22:
I also have this setup and would like the Deskpi install scripts to work with HAOS.

yoyojacky, 2024-01-30:
please refer to this url: https://community.home-assistant.io/t/deskpi-pro-active-cooling-and-front-usb-ports/265136

closed 2024-01-30 by yoyojacky