DeskPi-Team/deskpi #127: DeskPi Lite doesn't react on pwm_xxx to /dev/ttyUSB0

vzhestkov, 2022-07-07:
PWM control is not working on DeskPi Lite, while `echo power_off > /dev/ttyUSB0` leads to poweroff.
`dtoverlay=gpio-fan,gpiopin=14,temp=45000` works but switching from full speed to off almost instantly at 45 degrees

vzhestkov, 2022-07-12:
I think I found the root cause of the issue:
https://deskpi.com/collections/deskpi-lite/products/new-deskpi-lite-set-top-box-for-raspberry-pi-4 points to the wrong github repo, as Lite version of the case doesn't work with the drivers from this project, but works with https://github.com/DeskPi-Team/deskpi_v1

I think it's confusing the users.

yoyojacky, 2023-11-23:
> I think I found the root cause of the issue: https://deskpi.com/collections/deskpi-lite/products/new-deskpi-lite-set-top-box-for-raspberry-pi-4 points to the wrong github repo, as Lite version of [...]

thanks a lot , modify the url right way.

closed 2023-11-23 by yoyojacky