DeskPi-Team/deskpi #128: Ubuntu Server 22.04 power led keeps blue after halt

jzazo, 2022-07-31:
I have installed Ubuntu Server 22.04 and I also installed drivers for Deskpi Pro. I think fan is working, and secondary USBs are also working.

However, when I shutdown the rpi4, all services turn off but the blue led keeps been lit and it does not turn red. Before Ubuntu I was using Raspbian and it was working fine. Any idea why this happens? Thank you.

jzazo, 2022-08-13:
Hi again. Any help regarding automatic flipping of led to red after `sudo shutdown -h now`? Do I need to change the eeprom config? Does it happen with other distributions? Thanks.

jzazo, 2022-09-23:
Nobody can help? Thanks.

beforeyouknowit, 2022-12-13:
I'm also having this issue, on my Pi 4 / 4GB, under Ubuntu 22.04; pressing the power button on my DeskPi Pro does not trigger a safe shutdown. I've had a look at your various installation scripts, and I'm trying to make sense of the patterns used.

In the `/drivers/c/` folder, I see a `safecutoffpower.c` and it's compiled binary`safecutoffpower`. In the same folder, I see a `safecutoffpower64` binary but without its corresponding C file. Is this 64-bit-oriented C file missing?

Also, following your installation script patterns, in the `install-ubuntu-64.sh` script, (that I used for Ubuntu 22.04 (64-bit) ARM,) I don't see a similar pattern you used to compile the `pwmControlFan.c` file into the `pwmFanControl` binary, which might correspond to a `gcc` compilation step for the 64-bit `safecutoffpower64` binary, nor a similar step for compiling the `fanStop.c` file for 64-bit.

Could you please help us understand if we should be focusing on manually compiling (and copying into place) the binaries rendered from `fanStop.c` or missing `safecutoffpower64.c` ? Thanks!

yoyojacky, 2023-02-17:
> 

Hi there, 
the power led turn to red means that your system has been sent "POWER_OFF" signal via serial port to /dev/ttyUSB0 device , once you shut down your system , deskpi service will send the "power_off" signal to /dev/ttyUSB0 serial port , it will be detected by MCU on mainboard, so it will cut off the power from mainboard to raspberry pi.  it seems like the ubuntu_installation script dose not work properly, i am going to fix this issue ASAP.

yoyojacky, 2023-02-17:
> 

It is the same source code ,but compiled in both 32bit and 64bit system .

lfhoward, 2023-05-15:
Hello, the same issue still exists as of May 14, 2023. It is a fresh install of Ubuntu 22.04. Soft shutdown does not work and the power button stays blue after shutting down via the on-screen menu. Thanks for addressing this bug.

lfhoward, 2023-05-15:
I've been doing a little investigating, seeing where the requisite files are located for the safe shutdown service.

The necessary files seem to be: (which get installed by setup)
/usr/bin/fanstop 
/usr/lib/systemd/system/deskpi-safeshut.service
/etc/systemd/system/halt.target.wants/deskpi-safeshut.service
/etc/systemd/system/shutdown.target.wants/deskpi-safeshut.service
/etc/systemd/system/poweroff.target.wants/deskpi-safeshut.service

probably missing:
safeCutoffPower64  (located in ~/deskpi/drivers/c/ and could be moved to /usr/bin/, which would be where the pwmFanControl64 file goes.) This likely sends the ‘Power_Off’ signal to the serial port at /dev/ttyUSB0, which is the same serial used by the fan control. I suspect this because the code is probably derived from safecutoffpower.py in ~/deskpi/drivers/python/.

The deskpi-safeshut.service file, found in 4 locations above, looks like:

[Unit]
Description=DeskPi Safeshutdown Service
Conflicts=reboot.target
Before=halt.target shutdown.target poweroff.target
DefaultDependencies=no
[Service]
Type=oneshot
ExecStart=/usr/bin/sudo /usr/bin/fanStop
RemainAfterExit=yes
TimeoutSec=1
[Install]
WantedBy=halt.target shutdown.target poweroff.target

I wonder if the problem is in this line of code: ExecStart=/usr/bin/sudo /usr/bin/fanStop

I can see how it calls fanstop, but it doesn’t call safeCutoffPower64. The parallel line of code in deskpi.service is: ExecStart=sudo /usr/bin/pwmFanControl64

I do not know where the safeCutoffPower64 should be called in this line of code, whether before or after sudo and fanstop. Can anyone on here shed some light on how and when safeCutoffPower64 should be called inside of deskpi-safeshut.service?

lfhoward, 2023-05-15:
Partial Solution:

I checked in /usr/bin and safecutoffpower64 was not there.

I then copied safecutoffpower64 from ~/deskpi/drivers/c/ to /usr/bin/

I then edited /lib/systemd/system/deskpi-safeshut.service to run the safecutoffpower64 function after fanStop:

[Unit]
Description=DeskPi Safeshutdown Service
Conflicts=reboot.target
Before=halt.target shutdown.target poweroff.target
DefaultDependencies=no
[Service]
Type=oneshot
ExecStart=/usr/bin/sudo /usr/bin/fanStop /usr/bin/safecutoffpower64
RemainAfterExit=yes
TimeoutSec=1
[Install]
WantedBy=halt.target shutdown.target poweroff.target

Make sure the ExecStart line is edited in these 4 locations:
/usr/lib/systemd/system/deskpi-safeshut.service
/etc/systemd/system/halt.target.wants/deskpi-safeshut.service
/etc/systemd/system/shutdown.target.wants/deskpi-safeshut.service
/etc/systemd/system/poweroff.target.wants/deskpi-safeshut.service

sudo shutdown now 

The light should turn red a few seconds after shutdown.

After booting back up, I tried using the button (short press) to initiate a safe shutdown and still nothing happens. I am not sure how to get the button to initiate the shutdown process. The function /usr/bin/poweroff could be called somehow, but I am not sure how to assign that to pressing the button. However the light behind the button acts like it should now, and the power does not have to be cut manually after shutdown anymore.

Hope this helps someone!

lfhoward, 2023-05-15:
While looking at the drivers for an earlier version of DeskPi (Lite, V1), I noticed that there is a function for turning the Pi off when the power button is pressed twice. There does not seem to be such a file included in the DeskPi Pro (V2) drivers.

The earlier driver's code can be found here: [https://github.com/DeskPi-Team/deskpi_v1/blob/main/drivers/python/safe_shutdown.py](https://github.com/DeskPi-Team/deskpi_v1/blob/main/drivers/python/safe_shutdown.py)

It listens for a ‘poweroff’ signal from the MCU (button), and returns a power off signal to /dev/ttyUSB0 to run the safe shutdown service if it gets the message. I have not tried to run that old driver in my DeskPi Pro. I don’t know if the button will send the ‘poweroff’ (no underscore) signal properly, nor do I know if the shutdown script might need to be modified: the power off signal sent to /dev/ttyUSB0 in the old driver is the all lowercase 'power_off', but in the new DeskPi Pro drivers it is 'Power_Off'. I don’t know if the change to upper case letters matters.

cchulo, 2023-06-22:
are you at least able to run the fans, I'm having an issue with both the power button staying blue and fans not running

jzazo, 2023-06-22:
A PR has been merged that should have fixed the blue led staying on. Latest master should include the fix.

cchulo, 2023-06-22:
> A PR has been merged that should have fixed the blue led staying on. Latest master should include the fix.

it does not, I tried it on Ubuntu server/desktop 22.04, and raspberry pi OS all 64 bit, and on shutdown the blue LED stays on

I was able to get the fans to work on raspberry pi OS, but nothing else works. I am going to be RMA'ing it, I am disappointed by the state of these setup scripts

yoyojacky, 2023-07-05:
I am working on it, it will comming soon.

yoyojacky, 2024-07-05:
> it does not, I tried it on Ubuntu server/desktop 22.04, and raspberry pi OS all 64 bit, and on shutdown the blue LED stays on
>
> I was able to get the fans to work on raspberry pi OS, but nothing else [...]

It seems like your system did not send `power_off` signal to the MCU so that it will not cut off the power, and could you please check the `always on` switch on deskpi daughter board ?