DeskPi-Team/deskpi #132: Deskpi prov2 fan control not working

D-A-H-8-5, 2022-09-30:
Like the title says the fan control on my new deskpi pro v2 is not working.  install was succesful and after reboot this is what I get:
` deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Fri 2022-09-30 19:10:06 EDT; 1min 12s ago
    Process: 2369 ExecStart=/usr/bin/pwmFanControl64 & (code=killed, signal=TERM)
   Main PID: 2369 (code=killed, signal=TERM)
        CPU: 29ms

Sep 30 19:09:08 dah systemd[1]: Starting DeskPi Fan Control Service...
Sep 30 19:10:06 dah systemd[1]: deskpi.service: Main process exited, code=killed, status=15/TERM
Sep 30 19:10:06 dah systemd[1]: deskpi.service: Failed with result 'signal'.
Sep 30 19:10:06 dah systemd[1]: Stopped DeskPi Fan Control Service.
` 
here is the detail on my system:
`neofetch
            .-/+oossssoo+/-.               dah@dah 
        `:+ssssssssssssssssss+:`           ------- 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 22.04.1 LTS aarch64 
    .ossssssssssssssssssdMMMNysssso.       Host: Raspberry Pi 4 Model B Rev 1.5 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 5.15.0-1015-raspi 
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 10 mins 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 1836 (dpkg), 5 (flatpak), 16 (snap) 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 5.1.16 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 2560x1080 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   DE: GNOME 42.4 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   WM: Mutter 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   WM Theme: Adwaita 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Theme: Yaru-dark [GTK2/3] 
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/    Icons: Yaru [GTK2/3] 
  +sssssssssdmydMMMMMMMMddddyssssssss+     Terminal: gnome-terminal 
   /ssssssssssshdmNNNNmyNMMMMhssssss/      CPU: BCM2835 (4) @ 1.800GHz 
    .ossssssssssssssssssdMMMNysssso.       Memory: 1674MiB / 7806MiB 
      -+sssssssssssssssssyyyssss+-
        `:+ssssssssssssssssss+:`                                   
            .-/+oossssoo+/-.                                       
`

yoyojacky, 2023-02-17:
> Like the title says the fan control on my new deskpi pro v2 is not working. install was succesful and after reboot this is what I get: ` deskpi.service - DeskPi Fan Control Service Loaded: loaded [...]

Hi there, 
It seems like the binary file dose not work properly, i am going to rewrite the installation script, and i will send you the latest one ASAP. sorry for the trouble.

bebe1015, 2023-03-12:
Hi,
I have had the same issue using ubuntu
It was working with the raspian image.
Is the fix available for everyone?
thanks

yoyojacky, 2023-03-13:
> Hi, I have had the same issue using ubuntu It was working with the raspian image. Is the fix available for everyone? thanks

OK， working on it.

bebe1015, 2023-03-13:
> > Hi, I have had the same issue using ubuntu It was working with the raspian image. Is the fix available for everyone? thanks
> 
> OK， working on it.

Thank you very much.

jzazo, 2023-04-09:
How is the work coming along?

ddetton, 2023-04-18:
I used to have the fan control working with an earlier version of raspios 64 bit but now the service will not start.  I was trying to get pop-os fan control working which is not listed as a supported distro so I went back to raspios 64 and no go there either.  Is fan control working anywhere with current distros?

noanight, 2023-04-19:
@ddetton you may have a look at this [issue 142](https://github.com/DeskPi-Team/deskpi/issues/142)...i got it running with Ubuntu mate 64 bit.

yoyojacky, 2023-04-21:
> 

what version of your raspberry pi 4B？ V1.4 or V1.5 ? it seems like your raspberry pi 4B's USB-C port has some problem with it.

yoyojacky, 2023-04-21:
> 

fan spinning requires some conditions: 
1. Your OS can detect the serial port via USB-C port of Raspberry Pi, there may generate a serial port called "/dev/ttyUSB0" 
2. You need to send "PWMXXX“ to this serial port , XXX dedicates 000 to 100, it means the PWM dutycycle's value from 0 to 100,  for example, if you send "PWM075" means let the fan spinning at 75% speed. 
3. It can be drive in C language or Python language, if you want to use python script, you may need to install pyserial library to drive the serial port.

yoyojacky, 2024-05-13:
> seems like the binary file dose not work properly, i am going to rewrite the installation script, and i will send you the latest one ASAP. sorry for the trouble.

I have just create a brench on feature/bookworm and it under testing .

yoyojacky, 2024-07-05:
> How is the work coming along?

I have test it on UBUNTU 24.04 LTS but no luck, it seems like the dwc2 overlay is malfunction. I am trying other way to recognize the serial port on daugherboard of deskpi.

closed 2024-07-05 by yoyojacky