DeskPi-Team/deskpi #139: Automatic Fan Doesn't Work

bryantwaller, 2023-01-21:
Raspberry Pi 4 8gb
Raspberry Pi OS  64 Bit
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
Deskpi Pro V2

The fan doesn't work in automatic mode.  
When I boot from SD card with no SSD card in I can control fan in manual, but no Automatic. If I run the command deskpi-config and choose 7 - Cancel manual control and enable automatic fan control
If I install an SSD the fan doesn't work at all and the front USB ports don't work either. If I run the command deskpi-config and choose 1 - 4 nothing happens.

sudo systemctl status deskpi
● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-01-21 12:13:33 EST; 1h 49min ago
    Process: 1778 ExecStart=sudo /usr/bin/pwmFanControl (code=exited, status=1/FAILURE)
   Main PID: 1778 (code=exited, status=1/FAILURE)
        CPU: 32ms

Jan 21 12:13:33 raspberrypi systemd[1]: Started DeskPi PWM Control Fan Service.
Jan 21 12:13:33 raspberrypi sudo[1778]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
Jan 21 12:13:33 raspberrypi sudo[1778]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Jan 21 12:13:33 raspberrypi sudo[1778]: pam_unix(sudo:session): session closed for user root
Jan 21 12:13:33 raspberrypi sudo[1778]: sudo: unable to execute /usr/bin/pwmFanControl: No such file or directory
Jan 21 12:13:33 raspberrypi systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
Jan 21 12:13:33 raspberrypi systemd[1]: deskpi.service: Failed with result 'exit-code'.

kaimwe, 2023-01-21:
I just installed a Deskpi Pro V2 for the first time yesterday with the same Linux variant (SSD). After using the script 'install-raspios-64bit.sh' instead of 'install.sh' it worked for me (pwmFanControl64 is installed, not pwmFanControl). Possibly afterwards configure the automatic fan control again with the help of deskpi-config.

bryantwaller, 2023-01-22:
I did a clean install and was able to get it running, and configured.

but it seems to stop responding after 10 or 15 minutes, and I have to run the command deskpi-config and choose 7 to get it started again.

Is there a way to keep it running?

kaimwe, 2023-01-29:
Hmmm, seems to be the same for me, that the fan control stops working after some time. I currently only get to play around with the Pi from time to time, if I find out anything, I'll let you know here again.

kaimwe, 2023-01-31:
I solved the problem with the solution from closed issue #81 
(used the given python script and the modified deskpi.service)

bryantwaller, 2023-02-01:
> I solved the problem with the solution from closed issue #81 (used the given python script and the modified deskpi.service)

Thank you kaimwe. I will have to give it a try. I did move the rpi to another case... but this will definitely be used on my next pi.

yoyojacky, 2023-02-17:
>
 Hi there, it seems like you have installed 32bit version, could you please try to install 64bit driver for the fan control?

yoyojacky, 2023-02-17:
COMMAND=/usr/bin/pwmFanControl　this file is 32bit version , you can just download the pwmFanControl64 to same folder and reboot your raspberry pi, and try again. do not forget change the "COMMAND=/usr/bin/pwmFanControl64" in /lib/systemd/system/deskpi.service

yoyojacky, 2023-02-17:
> 
Did you notice the temperature of CPU when the fan stops?  if you use automatically control, it will control the fan according to the temperature of the CPU, if the CPU temperature is lower than the threshold you setting , it may not spinning anymore, so you may need to check if the /etc/deskpi.conf is available.

closed 2023-02-27 by yoyojacky