DeskPi-Team/deskpi #141: Dietpi 64 Fan Control

bitzerjd99, 2023-04-08:
How do I get this to work....

● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-04-08 16:05:16 EDT; 2min 12s ago
    Process: 615 ExecStart=sudo /usr/bin/pwmFanControl (code=exited, status=1/FAILURE)
   Main PID: 615 (code=exited, status=1/FAILURE)
        CPU: 32ms

Apr 08 16:05:16 OCTOPRINT02 systemd[1]: Started DeskPi PWM Control Fan Service.
Apr 08 16:05:16 OCTOPRINT02 sudo[615]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
Apr 08 16:05:16 OCTOPRINT02 sudo[615]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Apr 08 16:05:16 OCTOPRINT02 sudo[615]: pam_unix(sudo:session): session closed for user root
Apr 08 16:05:16 OCTOPRINT02 sudo[615]: sudo: unable to execute /usr/bin/pwmFanControl: No such file or directory
Apr 08 16:05:16 OCTOPRINT02 systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
Apr 08 16:05:16 OCTOPRINT02 systemd[1]: deskpi.service: Failed with result 'exit-code'.

I tried the Raspberry 64 script and can get the driver to load, but it crashes within a few minutes.

The file is where it says it should be...but I get the above error message.

Thanks!

bitzerjd99, 2023-04-08:
If I use pwmFanControl64, I get this...but the fan doesn't seem to be working

deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-04-08 17:12:42 EDT; 8s ago
   Main PID: 1577 (sudo)
      Tasks: 2 (limit: 4531)
        CPU: 20ms
     CGroup: /system.slice/deskpi.service
             ├─1577 sudo /usr/bin/pwmFanControl64 &
             └─1578 /usr/bin/pwmFanControl64 &

Apr 08 17:12:42 OCTOPRINT02 systemd[1]: Started DeskPi PWM Control Fan Service.
Apr 08 17:12:42 OCTOPRINT02 sudo[1577]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl64 &
Apr 08 17:12:42 OCTOPRINT02 sudo[1577]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)

My temp is sitting at 141 Degrees.  Dietpi is complaining it is way too hot.  The PI is doing very little so the temp should be lower.

I've tried changing the fan control to 100% all the time, but it doesn't do anything.

bitzerjd99, 2023-04-10:
pwmFanControl64 stopped working...it hangs on starting.  Is there any support here?

paulwratt, 2023-11-03:
if your repo is up to date now, you can look in the Ubuntu64 install script, and manually run the `gcc` command that generates and installs `/usr/bin/pwmFanControl64`.

There have been 2x PR's that fix bugs in `pwmControlFan.c` since this was posted

yoyojacky, 2024-07-05:
[image]

closed 2024-07-05 by yoyojacky