DeskPi-Team/deskpi #142: Segmentation Faults ubuntu 22.10 x64

tardisjacker, 2023-04-10:
I've got a deskpi pro with the Ubuntu latest 22.10 image installed booting via the USB SSD adapter that came as a part of the kit. I've pulled and installed the latest version and have issues with Segmentation Fault errors without any further information. It seems that these tools are not yet ready for 22.10.

1. Out of the gate I kept receiving errors about needing to change dtoverlay settings in /boot/configs.txt that were already enabled under `/boot/firmware/configs.txt` and that ttyUSB0 wasn't found. a test case for where the configs are may be appropriate

```
# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2,dr_mode=host
```
 
3.  while It appears that the patches found in https://github.com/DeskPi-Team/deskpi/issues/118#issue-1272238182 will likely help with issues related to the device naming especially when using any other USB serial devices; the issue on a system with only the deskpi seems related to a match found via brltty, with a fix found here: https://github.com/juliagoda/CH341SER/issues/18#issuecomment-1413273419
```
comment out this line:
ENV{PRODUCT}=="1a86/7523/*", ENV{BRLTTY_BRAILLE_DRIVER}="bm", GOTO="brltty_usb_run"
in this file:
/usr/lib/udev/rules.d/85-brltty.rules
Then reboot
```

4. Now that we've sorted the issues with /dev/ttyUSB0 via the brltty fix, running the command to start the service returns a segfault
```
me@deskpi:~$ sudo /usr/bin/pwmFanControl
Segmentation fault
```

Info provided by: `journalctl -xeu deskpi.service`
```
Apr 10 11:27:55 deskpi systemd[1]: Starting DeskPi PWM Control Fan Service...
░░ Subject: A start job for unit deskpi.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit deskpi.service has begun execution.
░░ 
░░ The job identifier is 206.
Apr 10 11:27:55 deskpi sudo[1598]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl &
Apr 10 11:27:55 deskpi sudo[1598]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Apr 10 11:27:56 deskpi sudo[1598]: pam_unix(sudo:session): session closed for user root
Apr 10 11:27:56 deskpi systemd[1]: deskpi.service: Main process exited, code=killed, status=11/SEGV
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ An ExecStart= process belonging to unit deskpi.service has exited.
░░ 
░░ The process' exit code is 'killed' and its exit status is 11.
Apr 10 11:27:56 deskpi systemd[1]: deskpi.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit deskpi.service has entered the 'failed' state with result 'signal'.
Apr 10 11:27:56 deskpi systemd[1]: Failed to start DeskPi PWM Control Fan Service.
░░ Subject: A start job for unit deskpi.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit deskpi.service has finished with a failure.
░░ 
░░ The job identifier is 206 and the job result is failed.
Apr 10 11:46:07 deskpi systemd[1]: Starting DeskPi PWM Control Fan Service...
░░ Subject: A start job for unit deskpi.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit deskpi.service has begun execution.
░░ 
░░ The job identifier is 3426.
Apr 10 11:46:07 deskpi sudo[2826]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl &
Apr 10 11:46:07 deskpi sudo[2826]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Apr 10 11:46:08 deskpi sudo[2826]: pam_unix(sudo:session): session closed for user root
Apr 10 11:46:08 deskpi systemd[1]: deskpi.service: Main process exited, code=killed, status=11/SEGV
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ An ExecStart= process belonging to unit deskpi.service has exited.
░░ 
░░ The process' exit code is 'killed' and its exit status is 11.
Apr 10 11:46:08 deskpi systemd[1]: deskpi.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit deskpi.service has entered the 'failed' state with result 'signal'.
Apr 10 11:46:08 deskpi systemd[1]: Failed to start DeskPi PWM Control Fan Service.
░░ Subject: A start job for unit deskpi.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit deskpi.service has finished with a failure.
░░ 
░░ The job identifier is 3426 and the job result is failed.
me@deskpi:~/deskpi$ 
```

and from `systemctl status deskpi.service`
```
jgraham@jaydon-deskpi:~/deskpi$ systemctl status deskpi.service 
× deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; preset: enabled)
     Active: failed (Result: signal) since Mon 2023-04-10 11:46:08 MST; 4min 24s ago
    Process: 2826 ExecStart=sudo /usr/bin/pwmFanControl & (code=killed, signal=SEGV)
   Main PID: 2826 (code=killed, signal=SEGV)
        CPU: 35ms

Apr 10 11:46:07 jaydon-deskpi systemd[1]: Starting DeskPi PWM Control Fan Service...
Apr 10 11:46:07 jaydon-deskpi sudo[2826]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl &
Apr 10 11:46:07 jaydon-deskpi sudo[2826]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Apr 10 11:46:08 jaydon-deskpi sudo[2826]: pam_unix(sudo:session): session closed for user root
Apr 10 11:46:08 jaydon-deskpi systemd[1]: deskpi.service: Main process exited, code=killed, status=11/SEGV
Apr 10 11:46:08 jaydon-deskpi systemd[1]: deskpi.service: Failed with result 'signal'.
Apr 10 11:46:08 jaydon-deskpi systemd[1]: Failed to start DeskPi PWM Control Fan Service.
```
I'll start looking into how the build is being done and if there are other issues in here that may have fixes via patches or changes to the systemd service portions but at this point the only hint I've got is a Segmentation Fault with return code 11 if anyone has any insight there

jzazo, 2023-04-10:
Are you using 64bits? I had some back and forths with 22.04, have you tried to install with these [changes](https://github.com/DeskPi-Team/deskpi/pull/140)?

tardisjacker, 2023-04-13:
I seem to have figured it out, due to the complications in my installation; for whatever reason `/etc/deskpi.conf
`
wasn't created when I installed or attempted re-installs with the script. I was able to use the `deskpi-config
` with the option to set my own curve which created the file that fixed my issue. I will attempt a clean re-image of my PI with the mentioned fix commenting out `ENV{PRODUCT}=="1a86/7523/*", ENV{BRLTTY_BRAILLE_DRIVER}="bm", GOTO="brltty_usb_run"
` In `/usr/lib/udev/rules.d/85-brltty.rules
`
and report back here if that's what it takes to install on 22.10

noanight, 2023-04-14:
@tardisjacker thx, ttyUSB0 is no here.

but calling:
sudo pwmControlFan returns
"Segmentation fault" too...

yoyojacky, 2023-04-17:
please check your OS platform, if you are using 64bit arch, please use pwmfancontrol64  it is a 64bit binary file,  pwmfancontrol is 32bit

yoyojacky, 2023-04-17:
> sudo /usr/bin/pwmFanControl

try to cope pwmFanControl64 to /usr/bin/ location and then execute :
> sudo /usr/bin/pwmFanControl64

noanight, 2023-04-17:
@yoyojacky i'm using mate 64bit

- pwmFanControl
modified install-ubuntu-64.sh to:
  ```shell
  sudo cp -rf $installationfolder/drivers/c/pwmFanControl64 /usr/bin/pwmFanControl
  ```
  running sudo /usr/bin/pwmFanControl
  No Segmentation Fault Error anymore.
  
  But why does the install script compile pwmControlFan.c
  gcc -o $installationfolder/drivers/c/pwmFanControl $installationfolder/drivers/c/pwmControlFan.c
  by compiling it, it should be 64Bit binary, so why an extra pwmFanControl64?

- What is with fanStop? Is it a 64 Bit executable?

- Testing the cooler with deskpi-config 
running deskpi-config with 100% -> cooler does nothing....

yoyojacky, 2024-07-05:
> I'm using mate 64 bit and have the same issue. Installed with install-ubuntu-64.sh and added the compilation of fanStop.c. There is no /dev/ttyUSB0 device listed with ls /dev/tty*
> 
> Why is /dev/ttyUSB0 not created? There is a /dev/ttyAMA0 device...

I have the same issure about this OS right now, it can not generate the '/dev/ttyUSB0' device and also makes the front USB out of work too, I am working on this and find that the dwc2 overlay seems out of work, I have open an issure for Ubuntu maintance team, but no response from them. ...

closed 2024-07-05 by yoyojacky