DeskPi-Team/deskpi #146: Cooling Fan Unresponsive in Ubuntu 22.04 on 8GB RPi 4

lfhoward, 2023-05-15:
Hello,

I installed DeskPi Pro on an 8 GB Raspberry Pi 4 running Ubuntu 22.04.02. I installed it using this method:

For Ubuntu 64bit OS
cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install-ubuntu-64.sh
sudo ./install-ubuntu-64.sh

The fan controller does not seem to work properly. The fan runs continuously at a slow speed and does not change speeds when I choose different settings in deskpi-config. 

I ran systemctl status deskpi.service to see what the problem was, and got this result:

lauren@lauren-pi:~$ systemctl status deskpi.service
× deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Mon 2023-05-15 01:27:55 EDT; 2min 39s ago
    Process: 1954 ExecStart=sudo /usr/bin/pwmFanControl (code=killed, signal=SEGV)
   Main PID: 1954 (code=killed, signal=SEGV)
        CPU: 49ms

May 15 01:27:55 lauren-pi systemd[1]: Started DeskPi PWM Control Fan Service.
May 15 01:27:55 lauren-pi sudo[1954]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
May 15 01:27:55 lauren-pi sudo[1954]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
May 15 01:27:55 lauren-pi sudo[1954]: pam_unix(sudo:session): session closed for user root
May 15 01:27:55 lauren-pi systemd[1]: deskpi.service: Main process exited, code=killed, status=11/SEGV
May 15 01:27:55 lauren-pi systemd[1]: deskpi.service: Failed with result 'signal'.

According to the result above, there is a signal error, which killed the process. 

In researching this problem, I found this written in the DeskPi Pro wiki:

“if you are using 64bit OS, The script to control the fan is in the drivers/c/ directory. The file suffix with 64 means 64bit, and the other one is a 32bit executable file.”

I checked in /usr/bin and pwmFanControl was there but not pwmFanControl64.

I then copied pwmFanControl64 from ~/deskpi/drivers/c/ to /usr/bin/

I then edited /lib/systemd/system/deskpi.service to reference the 64 bit driver instead of the 32 bit driver:

  GNU nano 6.2                                       /lib/systemd/system/deskpi.service                                                
[Unit]
Description=DeskPi PWM Control Fan Service
After=multi-user.target
[Service]
Type=simple
RemainAfterExit=no
ExecStart=sudo /usr/bin/pwmFanControl64
[Install]
WantedBy=multi-user.target

sudo reboot
After booting up, the fan still does not work properly, but this time the error was different.

systemctl status deskpi.service result with pwmFanControl64:

lauren@lauren-pi:~$ systemctl status deskpi.service
× deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2023-05-15 01:06:31 EDT; 10s ago
    Process: 4486 ExecStart=sudo /usr/bin/pwmFanControl64 (code=exited, status=1/FAILURE)
   Main PID: 4486 (code=exited, status=1/FAILURE)
        CPU: 22ms

May 15 01:06:31 lauren-pi systemd[1]: Started DeskPi PWM Control Fan Service.
May 15 01:06:31 lauren-pi sudo[4486]: sudo: /usr/bin/pwmFanControl64: command not found
May 15 01:06:31 lauren-pi systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
May 15 01:06:31 lauren-pi systemd[1]: deskpi.service: Failed with result 'exit-code'.

According to this result, the process failed because a command was not found, whereas last time there was a signal issue.

The behavior of the fan remains unchanged, spinning at a slow speed, regardless of which mode I choose using deskpi-config.. 

Please advise on what to check next. Thanks in advance.

lfhoward, 2023-05-15:
I was able to get it to change fan speeds by doing the following:

Installed vgencmd because it is called by /usr/bin/deskpi-config to measure_temp. 

Searched for DeskPi in files
Places where pwmFanControl are referenced, and where it can be changed to version 64
/lib/systemd/system/deskpi.service (I changed to pwmFanControl64)
/usr/lib/systemd/system/deskpi.service (already set to 64)
/etc/systemd/system/multi-user.target.wants/deskpi.service (already set to 64)

The deskpi-config command writes the fan speed to the file: /dev/ttyUSB0
This writing works. But it doesn't affect fan speed with what I've done so far.

The eventual solution came from TardisJacker on https://github.com/DeskPi-Team/deskpi/issues/142 

comment out this line:
ENV{PRODUCT}=="1a86/7523/*", ENV{BRLTTY_BRAILLE_DRIVER}="bm", GOTO="brltty_usb_run"
in this file:
/usr/lib/udev/rules.d/85-brltty.rules
Then reboot

After this, I was able to use deskpi-config to change the fan speeds. 
I was also able to change speeds manually at the command line using:
sudo sh -c "echo pwm_100 > '/dev/ttyUSB0'"
...where the 100 could be changed to anything from 000 to 100.

If this by itself doesn't solve the problem, be sure /lib/systemd/system/deskpi.service references pwmFanControl64, and be sure vcgencmd is installed.

Hope this helps someone, and thanks again to TardisJacker for identifying the key line that caused the conflict.

cchulo, 2023-06-22:
make sure to `chmod +x pwnFanControl64` or at least `chmod 500`

muckypaws, 2023-06-25:
Thank you, this solution worked perfectly on Ubuntu 23.04.  Been scratching my head on why this was occurring, even putting the driver in debug and single stepping through.

closed 2024-07-06 by yoyojacky