DeskPi-Team/deskpi #148: Fan control assumes only one ttyUSB device is present.

lmjabreu, 2023-06-16:
I've got a projector connected to my raspberry pi via a USB serial adaptor which gets mapped to ttyUSB0.

All of the DeskPi scripts assume only one ttyUSB device, ttyUSB0, and therefore fan control fails.

In addition to correcting the startup scripts (as described in #81 and many, many other issues), I've set the tty device to ttyUSB1 for the fan control to work.

The script should probably validate this assumption somehow, i.e. the fan tty device should be able to provide a response, so the script can iterate through ttyUSBn until it finds an appropriate fan control response for e.g.

Note: the cooling provided by this case, even with the fan + ICE block is absolutely shocking compared to the passive cooling provided by the Akasa GemPro (31ºC passive cooling, versus 41ºC with the fan at 50%, exactly same raspberry pi hardware and setup). I wonder if it's worth switching to a good passive cooling, as used in the Akasa Gem Pro, to get rid of all the fan control issues and hardware costs. 😬

yoyojacky, 2023-07-04:
This is open source, the code is provided to you, and you can change it according to your own needs. Generally, my suggestion is to debug through deskpi-config. If you need to automatically adjust the speed, you can choose to add the parameters of CPU temperature threshold and fan speed. .

lmjabreu, 2023-07-04:
@yoyojacky A fun OSS project isn’t the same as OSS code written to support a key component of a product sold on Amazon, for profit.

People who buy the raspberry pi case won’t necessarily know how to fix a bad assumption made by the developer (there’s never more than one ttyUSB device), nor should they.

closed 2023-07-04 by lmjabreu

yoyojacky, 2023-07-05:
> A fun OSS project isn’t the same as OSS code written to support a key component of a product sold on Amazon, for profit.
> 
> People who buy the raspberry pi case won’t necessarily know how to fix a bad assumption made by the developer (there’s never more than one ttyUSB device), nor should they.

Well, it is true that customers will never follow the instructions to operate the equipment. We never know how customers will use the equipment, we can only use the most common usage possibilities to guess the user's usage scenarios. If you have special needs, please contact us for discussion.