DeskPi-Team/deskpi #149: Cooler mounting copper Pillar order help

RobotsAreCrazy, 2023-06-23:
Can someone help me with what order and type of copper pillar  is used on which part of the back of the pi 4 as the manual had me slightly confused on that part, can someone take a picture and label what size Pillar is used please just something I can work from?

Thanks so much 👍

yoyojacky, 2023-07-04:
Hello Sir, 
you can fix the copper pillar which is shorter in front of the raspberry pi(near by the TF card adapter board) , and two of the long copper pillar to the back of the raspberry pi (near by the Ethernet Port and USB ports).

RobotsAreCrazy, 2023-07-04:
So 2x short on side of tfcard reader and longer ones on the ethernet side?

yoyojacky, 2024-05-13:
yes,that's right.

closed 2024-07-05 by yoyojacky