DeskPi-Team/deskpi #152: Raspberry bullseye arm64 front usb ports not enabled

mgrouch, 2023-07-20:
It seems the script no longer adds

`dtoverlay=dwc2,dr_mode=host`

into /boot/config.txt

paulwratt, 2023-11-03:
not so:
```
sudo sed -i '/dtoverlay=dwc2*/d' /boot/config.txt 
sudo sed -i '$a\dtoverlay=dwc2,dr_mode=host' /boot/config.txt 
```
that should delete _any_ entry already in there
then _append_ a new one to the end (`$a`

yoyojacky, 2024-05-13:
> It seems the script no longer adds
> 
> `dtoverlay=dwc2,dr_mode=host`
> 
> into /boot/config.txt

new OS has been change to /boot/firmware/config.txt 
so, just modify to:
```bash
sudo sh -c "sudo sed -i  '/dtoverlay=dwc2*/d' /boot/firmware/config.txt 
sudo sh -c "sudo sed -i  '\$a\dtoverlay=dwc2,dr_mode=host' /boot/firmware/config.txt 
```
and do remember reboot your Raspberry Pi to take effect

closed 2024-05-13 by yoyojacky