DeskPi-Team/deskpi #153: pwmFanControl64 is opening serial port /dev/ttyUSB0

mgrouch, 2023-07-21:
ttyUSB0 is FTDI USB to serial adapter which has nothing to do with deskpi pro!
Absolutely unrelated device.

````

lrwxrwxrwx  1 root root          7 Jul 21 01:33 /dev/ttyLYS_ftdi_AO009078 -> ttyUSB0

user@lysmarine:~ $ sudo lsof /dev/ttyUSB0
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse.portal file system /run/user/1000/doc
      Output information may be incomplete.
COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
pwmFanCon 1509 root    3u   CHR  188,0      0t0  442 /dev/ttyUSB0

````

f-io, 2023-09-29:
had the same problem, in my case ttyUSB0 is an USB-> serial -> EnOcean device. Please update your driver to check first and not just hardcode the port. At lease give a warning for example if `ls /dev/ttyUSB*` has more than one result, most likely the fan will not work in this case.

paulwratt, 2023-11-03:
its because `deskpi-config`  has this at line 14:
```
serial_port='/dev/ttyUSB0'
```

I dont have any other USB TTY devices, so someone else will have to do some testing and `/sys/class/` lookups to see where the DeskPi TTY device ends up when you have another USB TTY device plugged in

HINT:
on RPi the USB devices are assigned from _top-down, left -to-right_ , which corresponds to there USB ID numbers (lowest to highest). `/dev/ttyUSB0` _only_ gets assigned to the DeskPi if no other USB TTY devices have already been registered with the stack.

- If you have 1 USB TTY device plugged into either of the USB2 ports, it will get assigned `/dev/ttyUSB0` _before_ the DeskPi , which means the DeskPi will have `/dev/ttyUSB1`.
- if you have 1 USB TTY device plugged into the "other" USB3 port, that will get `/dev/ttyUSB1` and DeskPi will still be on `/dev/ttyUSB0`
- if you have a hub on any of the USB2 ports, and multiple USB TTY devices, the DeskPi will get the _last_ `/dev/ttyUSB?` device

f-io, 2023-11-15:
Can we expect any fix on this? @yoyojacky 

It looks like the USB port you are utilizing is always the last one during initialization. 
I've tried the front USB (2.0), the back (2.0) and the second 3.0 USB port.

```
# dmesg | grep tty
[   11.853373] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0
[   14.987633] usb 3-1.4: ch341-uart converter now attached to ttyUSB1
```

f-io, 2023-11-15:
@yoyojacky possible solution, use udev rules. 

In this case:
```
#!/bin/bash

cat << EOF > /etc/udev/rules.d/10-local.rules 
ACTION=="add", ATTRS{idVendor}=="1d6b", ATTRS{idProduct}=="0002", SYMLINK+="deskPiPro"
EOF
```
The system will create a symlink `/dev/deskPiPro`. Update your pwmControlFan.c with `serial_port='/dev/deskPiPro'` and it will be independent.

yoyojacky, 2024-02-26:
yes, udev rule is better, I will try it on my test device, and update the script.

yoyojacky, 2024-05-13:
@f-io I've tried this, but and adding a udev rule file to /etc/udev/rules.d/10-deskpi.rules
and it can generate a symbolic link to /dev/ttyUSB0, like aliase name, but once I am trying to operate this device : /dev/deskPiPro 
it will not work properly. 
for example :  echo "pwm_100" > /dev/deskPiPro  
it will not trigger the fan spinning, but if i send the same command to /dev/ttyUSB0, it works fine .
and i have change the owner of the /dev/deskPiPro to current user by using chown command and chmod command .

yoyojacky, 2024-07-05:
> had the same problem, in my case ttyUSB0 is an USB-> serial -> EnOcean device. Please update your driver to check first and not just hardcode the port.

Did you connect other USB device on your raspberry pi  or just deskpi pro?

closed 2024-07-05 by yoyojacky