DeskPi-Team/deskpi #154: fan will not restart automatically after lock/sleep - ubuntu 22.04 (64 bit)

lfhoward, 2023-08-11:
I don't know if anyone knows how to make the deskpi fan restart when the computer wakes up from sleep or lock mode? I have to go to the terminal and type deskpi-config and re-set it up to get it running again. This is on Ubuntu 22.04, Pi-4 8 GB, 64 bit. Thanks in advance.

lfhoward, 2023-08-15:
Hi DeskPi users,

I and other users have found that the DeskPi fan service stops every 17-18 minutes (not only after waking up from lock mode, but all the time). Here is that thread:

[https://github.com/DeskPi-Team/deskpi/issues/81](https://github.com/DeskPi-Team/deskpi/issues/81) 

The workaround I used to keep the fan running is to add this line to the crontab and restart the service every 15 minutes:

*/15 * * * * systemctl restart deskpi

yoyojacky, 2024-05-22:
execute following command in terminal :
```bash
sudo systemctl restart deskpi.service 
```

closed 2024-05-22 by yoyojacky