DeskPi-Team/deskpi #156: White blinking LED

LPDaVinci, 2023-09-09:
Is there any choice to disable the white blinking LED on the deskpi case, when using a sata ssd?

gogreenpower, 2023-12-21:
I second this, very annoying

inspector71, 2023-12-23:
Doesn't bother me. I like it. However I can imagine situations where it could be annoying. An option to disable would be sensible.

gogreenpower, 2023-12-23:
Would de-soldering the led stop anything from working?

inspector71, 2023-12-24:
> Would de-soldering the led stop anything from working?

Wow, that's annoying it is? I find probably the simplest choice is a piece of black electrical tape :)

gogreenpower, 2023-12-24:
That's what I'm currently using but I like permanent solutions. I have gone to great lengths to reduce all electronic standby/on lights. Can't stand them.

yoyojacky, 2024-05-13:
I don't recommend to de-soldering it, it is a status indicator to check if the ssd is working properly, it will blink once the ssd is writing data or system is synchronizing... if you don't like it, just put a black tape on it. LoL

closed 2024-05-13 by yoyojacky