DeskPi-Team/deskpi #157: Doesnt work with new RaspberryPi OS (bookworm) just released

rbnpi, 2023-10-13:
A 64bit OS based on Debian Bookworm has just been released. This is partly to accommodate the new Raspberry Pi 5, but is also inteded to work with existing PI like the Pi4. I don't think the Pi5 will fit the case as the layout has changed, but I think you should produce a version that will work with existing Pi4 running this new OS. The Auto Power off doesn;t work, and the extra USB sockets are not recognised.
You can download the new OS from https://www.raspberrypi.com/software/operating-systems/
Note it is the 64bit version, There is a new 32bit one as well.

bobofruit1970, 2023-10-15:
We need A 64bit OS based version for Debian Bookworm.

theoak222, 2023-10-15:
I would like to add my voice to those interested in a working DeskPi Pro install script for RPi4B running new 64bit Raspberry Pi OS based on Debian Bookworm. The current script does not work with this new OS.

paulwratt, 2023-11-03:
> The current script does not work with this new OS.

look at the script `install-raspios-64bit.sh` and compare it to `install-ubuntu-64.sh`, you should be able to figure it out

Yawning, 2023-11-03:
For what it's worth, I rage-coded https://github.com/Yawning/deskpi-ctl as an alternative to this package, for both manual and automated fan control, and safe poweroff.

It should build and run on just about any Linux variant supported by the raspi 4, however everything was only tested on 64-bit Arch (btw).

bobofruit1970, 2023-11-03:
It's a pity that no one knows how to make the original services work with Debian Bookworm.

@Yawning could ask you to write in a few steps what commands to use to install your version.
I am not a programmer, but an ordinary user with very basic knowledge.

rbnpi, 2023-11-04:
There doesn't appear to be a port /dev/ttyUSB) showing on this OS. I tried running one of the scripts and it couldn't find it. Not sure why. The dwc2 overlay has got an entry to enable it in config.txt and the systemctl services all seem to be installed OK.
Nothing working though. Front USB ports not working and none of the scripts can communicate.
There is a section just before `dtoverlay=dwc.dr,dr_mode=host` which talks about disabling `otpg_mode=1` if DWC2 is required. I looked at a previous working OS configured for deskpi and it wasn't commented there so maybe you don't need to do that. I've tried it both ways.
Instead of ttyUSB0 there is a usb folder  in /dev with a single entry hiddev0 in it which isn't present on previous OS.
That's about as far as I've got.

CAN ANYONE FROM DeskPi Team PLEASE RESPOND?

Yawning, 2023-11-05:
Note: I haven't looked at the contents of the official repo (ie: this repo), or their scripts in a long time, so the automatic install script might be doing something weird.  I rewrote the tooling for a reason.

> There doesn't appear to be a port /dev/ttyUSB) showing on this OS.

Both the front USB ports, and the USB->Serial device work just fine on a fresh install of "Raspberry Pi OS (64-bit)" as long as the appropriate line is correctly added to `config.txt`, where the appropriate line is `dtoverlay=dwc2,dr_mode=host` (the very beginning of the file is fine).

bobofruit1970, 2023-11-05:
@Yawning 
How to install Your vesion?

rbnpi, 2023-11-05:
Hmm. I have `dtoverlay=dwc2,dr_mode=host` added in (in fact it is added in by the install script OK) but I still don't get the front usb. Works OK on the previous OS. You sure you are testing the latest bookworm based OS released 10 October 2023 and running under Wayland?

yoyojacky, 2023-11-06:
> There doesn't appear to be a port /dev/ttyUSB) showing on this OS. I tried running one of the scripts and it couldn't find it. Not sure why. The dwc2 overlay has got an entry to enable it in [...]
we are working at it, we have tried to test the latest version of Raspberry Pi OS and will give a solution ASAP.

theoak222, 2023-11-06:
Thank you yoyojacky!!!

Looking forward to new script.

rbnpi, 2023-11-07:
Yes thanks for looking into it yoyojacky

yoyojacky, 2023-11-09:
> Hmm. I have `dtoverlay=dwc2,dr_mode=host` added in (in fact it is added in by the install script OK) but I still don't get the front usb. Works OK on the previous OS. You sure you are testing the latest bookworm based OS released 10 October 2023 and running under Wayland?

did you reboot your raspberry pi after adding `dtoverlay=dwc2,dr_mode=host` ?

yoyojacky, 2023-11-09:
> A 64bit OS based on Debian Bookworm has just been released. This is partly to accommodate the new Raspberry Pi 5, but is also inteded to work with existing PI like the Pi4. I don't think the Pi5 [...]

Yes, we are rewrite the installation script and working on testing stuff. once it  done, the installation script will be updated ASAP.

yoyojacky, 2023-11-09:
> Thank you yoyojacky!!!
> 
> Looking forward to new script.

you are welcome, i am working on new script testing part, it will fit for bookworm 64bit  when i test it out.

rbnpi, 2023-11-09:
yoyojacky Yes I did reboot after adding the dwc2 line. I have several SC cards with Buster osc64 bit and desktop pro works perfectly with them so the hardware seems to be fine.  I also have some with the latest bookworm 64 bit OS and nothing works with them. Hope the script redevelopment goes well.

juanejot, 2023-11-14:
I purchased & have been happy with the DeskPi Pro until now. If the hardware design (including the portion that DeskPi is responsible for) is such that it needs a fan, the software by the company that sold the case needs to be updated to continue to do so. On bullseye aarch64, it worked. On bookworm aarch64, it does not.
A product your customer purchased should not expire because software is upgradable. You can be sure that if enough owner’s Raspberry Pis fry because your software is not turning the fan on to cool them, there will be market implications, at the very least.

rbnpi, 2023-11-15:
Just discovered there is an issue with dwct2 overlay which is down to Raspberry Pi
see this issue https://github.com/raspberrypi/bookworm-feedback/issues/77#issuecomment-1763386899

theoak222, 2023-11-18:
Just a question. Both the Github issue # 77 (firmware) and RPi forum posts (mentioned above) are RPi5 related. DeskPi Pro scripts are for RPi4. Will the work being done to fix dwc2 overlay on RPi5 carry over to the RPi4? Will the upcoming firmware released be compatible with both Pi4s and Pi5s?

paulwratt, 2023-11-18:
@juanejot  or you could _just read the script yourself_

Note to anyone else:
1) if you read the _fan speed script_ you will realize you dont **_need_** to set it with any PWM binary, you can just output a _value_ to the correct _USB tty_ device (usually 0).
2) your RPi4 will _throttle_ if it get too hot (as opposed to "burning up").
3) you _can_ use the Ice Tower coolers _without_ a fan.

theoak222, 2023-11-19:
I am actually interested in ALL features of DeskPi Pro working, IE: front USB ports, fan, and power switch. Yoyojacky seems to be working on this. This was not a "complaint" but  I did not want yoyojacky to stop working on new script due to thinking this is a Pi firmware issue (outside of his area of expertise). Not sure why you object to questions being asked??? 

If you have the answer to making all features of DeskPi Pro work with Bookworm version of RPiOS, please share. Otherwise I would appreciate yoyojacky working on the revised script.

yoyojacky, 2023-12-08:
> yoyojacky Yes I did reboot after adding the dwc2 line. I have several SC cards with Buster osc64 bit and desktop pro works perfectly with them so the hardware seems to be fine. I also have some with the latest bookworm 64 bit OS and nothing works with them. Hope the script redevelopment goes well.

ok, I have test it out on Raspberry Pi OS (64bit) bookworm, and trying to modify the install script to finish the job automatically, but it still has some issue on installation. so please wait for a while.

yoyojacky, 2023-12-08:
Acturally, it just a few  steps will make it work on bookworm. 
1.  Adding "dtoverlay=dwc2,dr_mode=host  to /boot/config.txt or /boot/firmware/config.txt file and save it , reboot raspberry pi.
2.  Download the repository from github.com/Deskpi-Team/deskpi.git  to local folder. such as /home/pi/deskpi or just /tmp/deskpi. 
3. Navigate to  deskpi/drivers/c/ folder and execute make clean, and then remake binary files : 
make 
4. But the installation script is not available for right now, I am working on this shell script to fit for more OS, such as batocera, ubuntu mate ,ubuntu 64bit and so on , it may take time, if just for Raspberry Pi OS 64bit bookworm, that will be easy.  sorry for my slow response, I have to test it more times to make sure it works properly .

theoak222, 2023-12-09:
Thank you so much for your work!! Looking forward to finished drivers and script!

rbnpi, 2023-12-09:
Now that RaspberryPI have tweaked the dwc2 overlay that works on an updated Raspberry Pi Bookworm OS, and I can now use the front usb ports on both Pi4 and Pi5. However, I still can see any port `/dev/USB0` on either Pi4 or Pi5 running Raspberry Pi OS Bookworm (up to date). This means that I can't access anything serially. If I switch back to Buster 64bit it all works fine on the Pi4 and the port is visible. Does anyone know how to get the port active? I've googled this but can't find any solution.

kidekoop2016, 2024-01-11:
Hi again,
Is there any new version of driver for Raspbian OS 64bit (bookworm).
We are still waiting for it.
Fan doesn't work.
Frontal USBs doesn't work.
There is no activity.
We would appreciate any progress or feedback.
Regards.

rjwill10, 2024-01-15:
I have the same issue. Thinking of just installing a different fan and connecting directly to the GPIO pins until the issue is resolved. I assume a new script will come soon, but the last update was over a month ago. 

The GeeekPi Armor Lite Heatsink with PWM will work with both the GPIO and the deskpi board.

kidekoop2016, 2024-01-19:
We have been waiting for a solution for a long time without a response.
It seems to me a lack of professionalism and no maintenance of the software.
You market a box that is not compatible with the latest operating systems.
I've had the project on hold for more than two months.
In December you responded that you would try to resolve it as soon as possible.
Please, we need a solution or else I will switch to competing boxes like:
Argon One.
At least deign to write a feedback explaining the progression.
all the best

rjwill10, 2024-01-27:
> I have the same issue. Thinking of just installing a different fan and connecting directly to the GPIO pins until the issue is resolved. I assume a new script will come soon, but the last update was [...]

I changed my fan to the one mentioned.. I also ran the manual updated given by the developer (see below)  and it works. If you can install deskpi-config, then the below is only a minor additional step:

#assumes you have already installed deskpi-config

# 1 - Edit config.txt to Enable dwc2 Overlay
sudo nano /boot/firmware/config.txt

# 2 - Following to end of file and save
dtoverlay=dwc2,dr_mode=host

# Reboot Raspberry Pi (optional)
sudo reboot

# 3 - Navigate to the drivers/c/ Folder and Compile
cd deskpi/drivers/c

# 4 - run
make clean

# 5 - run
make

# 6 - Control fan
cd ~
deskpi-config #set how you want

theoak222, 2024-01-28:
Using yoyojacky's instructions (posted Dec 8) I now have front USB ports working, power switch working and fan working. Happy camper here! 

Looking forward to finished script to make installation easier for those who may find those instructions hard to implement.

Thanks yoyojacky!!

kidekoop2016, 2024-02-25:
> Using yoyojacky's instructions (posted Dec 8) I now have front USB ports working, power switch working and fan working. Happy camper here!
> 
> Looking forward to finished script to make installation easier for those who may find those instructions hard to implement.
> 
> Thanks yoyojacky!!
> 
> EDIT: Also, there _**WAS**_ something involving dwc2 overlay that had to be fixed by Bookworm updates.

Dear theoak222 and yoyojacky, thank you¡
Using yoyojacky's instructions (posted Dec 8), I get frontal USB working.
But fan doesnt work. **Coul you explain us how to get working please?**

I did:

1. Adding "dtoverlay=dwc2,dr_mode=host to /boot/config.txt or /boot/firmware/config.txt file and save it , reboot raspberry pi.
2. Download the repository from github.com/Deskpi-Team/deskpi.git to local folder. such as /home/pi/deskpi or just /tmp/deskpi.
3. Navigate to deskpi/drivers/c/ folder and execute make clean, and then remake binary files: make
5.  Frontal USB working, but fan not
6. Install script as describer here: 
cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install-raspios-64bit.sh
sudo ./install-raspios-64bit.sh
make.
7. Fan doesnt workInstall script as describer here: 

You said you get fan working. How?
Thanks in advance

PD: Yoyo, any script update?

yoyojacky, 2024-02-26:
> o make installation easier for those who may find those instructions hard to implement.
> 
> Thanks yoyojacky!!

you are welcome bro, have a nice day!

yoyojacky, 2024-02-26:
> > Using yoyojacky's instructions (posted Dec 8) I now have front USB ports working, power switch working and fan working. Happy camper here!
> > Looking forward to finished script to make installation easier for those who may find those instructions hard to implement.
> > Thanks yoyojacky!!
> > EDIT: Also, there _**WAS**_ something involving dwc2 overlay that had to be fixed by Bookworm updates.
> 
> Dear theoak222 and yoyojacky, thank you¡ Using yoyojacky's instructions (posted Dec 8), I get frontal USB working. But fan doesnt work. **Could you explain us how you to get the fan working, please?**
> 
> I did:
> 
> 1. Adding "dtoverlay=dwc2,dr_mode=host to /boot/config.txt or /boot/firmware/config.txt file and save it , reboot raspberry pi.
> 2. Download the repository from github.com/Deskpi-Team/deskpi.git to local folder. such as /home/pi/deskpi or just /tmp/deskpi.
> 3. Navigate to deskpi/drivers/c/ folder and execute make clean, and then remake binary files: make
> 4. Frontal USB working, but fan not
> 5. Install script as describer here:
>    cd ~
>    git clone https://github.com/DeskPi-Team/deskpi.git
>    cd ~/deskpi/
>    chmod +x install-raspios-64bit.sh
>    sudo ./install-raspios-64bit.sh
>    make.
> 6. Fan doesnt work
> 
> You said you get fan working. How? Thanks in advance
> 
> PD: Yoyo, any script update?

No, We are working on new script which can detect the current OS version and choose a right configuration so that everybody who  has the deskpi pro want to install the script will be easy and we can do trouble shooting easily.  Due to too many new products need to be test, so this script test still on processing.. once it done ,you will be the first one to know, and wish you a fabulous day!！

kidekoop2016, 2024-03-24:
> No, We are working on new script which can detect the current OS version and choose a right configuration so that everybody who has the deskpi pro want to install the script will be easy and we can [...]

Dear friend,
Weather is becoming warmer and the Pi is gettin more and more temperature.
We are waiting for the script more than 4 months.
When do wou think it will be ready?
It is important.
Let us know.
regards¡

yoyojacky, 2024-04-09:
> Dear friend, Weather is becoming warmer and the Pi is gettin more and more temperature. We are waiting for the script more than 4 months. When do wou think it will be ready? It is important. Let us [...]

OK ,still in progress...

yoyojacky, 2024-05-13:
Hi bros,
I have already upload a new brench : feature/bookworm 
It can works on Raspberry Pi OS 64bit boomworm - which released at 2024-03-15

bobofruit1970, 2024-05-13:
In my config it does not work as it should.
USB works
manual fan speed sterring work
auto speed does not work

yoyojacky, 2024-05-15:
> In my config it does not work as it should.
> 
> 1. USB works
> 2. manual fan speed sterring work
> 3. Auto speed does nothing. Fan works with last manually set speed. I have checked config file /etc/deskpi.conf. Config file is ok.

> In my config it does not work as it should.
> 
> 1. USB works
> 2. manual fan speed sterring work
> 3. Auto speed does nothing. Fan works with last manually set speed. I have checked config file /etc/deskpi.conf. Config file is ok.

auto speed can not work properly ? Let me check it again .

joeyjoejoe503, 2024-05-18:
i found a solution last night that worked. For the deskpi-config just navigate to the .sh file under drivers section and copy the raw code. then in your terminal do the following:

sudo nano deskpi-config.sh

paste the code you copied from github

ctrl+x, enter and ctrl+o to exit

chmod +x install-deskpi-config.sh
sudo install-deskpi-config.sh

the process is the same for the fan control software. navigate to the .sh file under OS section and copy the raw code. then in your terminal do the following:

sudo nano deskpi-fan-control.sh

paste the code you copied from github

ctrl+x, enter and ctrl+o to exit

chmod +x install-deskpi-fan-control.sh
sudo install-deskpi-fan-control.sh

just input deskpi-config in your terminal and you will have access to the full fan control. i hope this helps.

bobofruit1970, 2024-05-19:
Sorry. I do not understand what You mean saying:
1. "just navigate to the .sh file under drivers section and copy the raw code"
2. "navigate to the .sh file under OS section and copy the raw code"

I do not know from where at github should I copy it. There are thousands .sh files at github

Can You point what .sh file You mean?

joeyjoejoe503, 2024-05-19:
> Sorry. I do not understand what You mean saying:
> 1. "just navigate to the .sh file under drivers section and copy the raw code"
> 2. "navigate to the .sh file under OS section and copy the raw code" [...]

The drivers folder is located on the deskpi GitHub where the original instructions are. Same for the OS.

bobofruit1970, 2024-05-20:
> The drivers folder is located on the deskpi GitHub where the original instructions are. Same for the OS.

There is no .sh file in DRIVERS dir.
So I still don't know what to coopy and from which file.
I can't find any directory named OS there either.
I'm sorry that you weren't able to provide the full name of these two files and the paths to them. I looked for these files for several hours and wondered what you might have meant. I'm giving up.

joeyjoejoe503, 2024-05-20:
> > > Sorry. I do not understand what You mean saying:
> > > 
> > > 1. "just navigate to the .sh file under drivers section and copy the raw code"
> > > 2. "navigate to the .sh file under OS section and copy the raw code"
> > > 
> > > I do not know from where at github should I copy it. There are thousands .sh files at github
> > > Can You point what .sh file You mean?
> > 
> > The drivers folder is located on the deskpi GitHub where the original instructions are. Same for the OS.
> 
> There is no .sh file in DRIVERS dir.
> So I still don't know what to copy and from which file.
> I can't find any directory named OS there either.
> I'm sorry that you weren't able to provide the full name of these two files and the paths to them. I looked for these files for several hours and wondered what you might have meant. I'm giving up.

The driver for the fan control is at the following:

https://github.com/DeskPi-Team/deskpi/blob/master/drivers/deskpi-config

What version of Linux are you using? That way I can send you the exact link since you can't find it.

yoyojacky, 2024-05-20:
> There is no .sh file in DRIVERS dir. So I still don't know what to copy and from which file. I can't find any directory named OS there either. I'm sorry that you weren't able to provide the full [...]

DRIVERS dir is contains the source code and compiled binary file, and makefile, you can compile it according to your system platform version by using `make` command, it is a open source code, and you can change it if you like.

yoyojacky, 2024-05-20:
> The driver for the fan control is at the following:
>
> https://github.com/DeskPi-Team/deskpi/blob/master/drivers/deskpi-config
>
> What version of Linux are you using? That way I can send you the exact [...]

yes, deskpi-config is just a shell script to call the function which copied from drivers/c/ folder, there are some souce code in side the folder, i am create a new brench and test it on several raspberry pi 4B platform on Raspberry Pi OS 64bit (bookworm). since the OS has been updated from bullseye to bookworm
If you have any question please discuss in this thread .

bobofruit1970, 2024-05-20:
Again from beginning.

1. In this discussion, as the title suggests, we are writing about the not working installation script and driver for OS bookworm. I have raspberry  PI 8GB and bookworm OS 64bit and I am verifying the operation of the installation script specially developed for bookworm by @yoyojacky. 

Installation script from directory:
https://github.com/DeskPi-Team/deskpi/tree/feature/bookworm/installation

As @yoyojacky  wrote:

> Hi bros, I have already upload a new branch : feature/bookworm It can works on Raspberry Pi OS 64bit bookworm - which released at 2024-03-15

2. I have tried to install it. After installation in my config everything worked except "auto fan speed feature".

3. @joeyjoejoe503 wrote:

> i found a solution last night that worked. For the deskpi-config just navigate to the .sh file under drivers section and copy the raw code. then in your terminal do the following:
>
> sudo nano [...]

4. Now I am at this point. 
@joeyjoejoe503 has found solution to make it work after installation from new script dedicated to bookworm OS.
I am kindly asking for a path to two files mentioned by @joeyjoejoe503 in his solution:
a) first file from DRIVERS directory
b) second file in OS directory. There is no such directory????

joeyjoejoe503, 2024-05-20:
No the installation script did not work for me, so I manually installed with the files that are available on the desk pi GitHub and it worked.

kidekoop2016, 2024-05-20:
Hi,
Thank you in advance. After 6 months seems that there are new.

I tried:
cd ~/deskpi/RaspberryPiOS/64bit
sudo chmod +x install-raspios-64bit.sh
sudo ./install-raspios-64bit.sh               
Error: 
Initializing functions....
./install-raspios-64bit.sh: línea 120: EOF inesperado mientras se buscaba un `"' coincidente

Could someboby write here the synthaxis with the exsact steps to proceed with the installation, please?

I have the old installation with the old script.
Thanks men

joeyjoejoe503, 2024-05-20:
> Hi,
> Thank you in advance. After 6 months seems that there are new.
>
> I tried:
> cd ~/deskpi/RaspberryPiOS/64bit
> sudo chmod +x install-raspios-64bit.sh
> sudo ./install-raspios-64bit.sh [...]

That's the same error I was getting which is why I chose to do it manually

kidekoop2016, 2024-05-26:
Ok¡
It is very confuse for me.
Coul you write here the steps, just writting the code to execute in terminal, please ?
Thanks

joeyjoejoe503, 2024-05-26:
> yes, deskpi-config is just a shell script to call the function which copied from drivers/c/ folder, there are some souce code in side the folder, i am create a new brench and test it on several [...]

I found an error in the install-raspios-64bit.sh file. On line 43 there is a " after sudo sh -c that needs to be removed. Once I removed it the install scripts worked

bobofruit1970, 2024-05-26:
@joeyjoejoe503 

There is directory dedicated to bookworm 
https://github.com/DeskPi-Team/deskpi/tree/feature/bookworm/installation

and  there is README file in it. This readme file says:
"Installation script for Raspberry Pi OS 64bit bookworm"

There is no such script file:
install-raspios-64bit.sh

There is install.sh file

About what installation script You write? For Buster OS 64bit?

joeyjoejoe503, 2024-05-26:
> @joeyjoejoe503
>
> There is directory dedicated to bookworm
> https://github.com/DeskPi-Team/deskpi/tree/feature/bookworm/installation
>
> and  there is README file in it. This readme file says: [...]

https://github.com/DeskPi-Team/deskpi/blob/master/RaspberryPiOS/install-raspios-64bit.sh

bobofruit1970, 2024-05-26:
Thank You @joeyjoejoe503.
So You work with a script for the previous Buster OS not that for Bookworm OS.

joeyjoejoe503, 2024-05-26:
It's the only one I could get to work. I'll post a complete guide on my page later tonight, I am currently away from my PC and doing it on mobile would make me insane.

kidekoop2016, 2024-05-26:
Hi,
Effectively. I hace instaled it using

git clone -b feature/bookworm https://github.com/deskpi-team/deskpi
cd deskpi/installation/
sudo ./install.sh 
------------------------------------------------------------------------------
``
Clonando en 'deskpi'...
remote: Enumerating objects: 703, done.
remote: Counting objects: 100% (276/276), done.
remote: Compressing objects: 100% (131/131), done.
remote: Total 703 (delta 174), reused 204 (delta 145), pack-reused 427
Recibiendo objetos: 100% (703/703), 10.13 MiB | 5.04 MiB/s, listo.
Resolviendo deltas: 100% (345/345), listo.
root@raspberrypi:/home/pi2/Downloads/deskpi-2024-2# cd deskpi/installation/
root@raspberrypi:/home/pi2/Downloads/deskpi-2024-2/deskpi/installation# sudo ./install.sh
Initializing functions....
Obj:1 http://archive.raspberrypi.com/debian bookworm InRelease
Obj:2 http://deb.debian.org/debian bookworm InRelease
Obj:3 https://download.docker.com/linux/debian bookworm InRelease
Obj:4 http://deb.debian.org/debian-security bookworm-security InRelease
Obj:5 http://deb.debian.org/debian bookworm-updates InRelease
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias... Hecho
Leyendo la información de estado... Hecho
Todos los paquetes están actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias... Hecho
Leyendo la información de estado... Hecho
Los paquetes indicados a continuación se instalaron de forma automática y ya no son necesarios.
  libraspberrypi0 libwpe-1.0-1 libwpebackend-fdo-1.0-1 linux-headers-6.1.0-rpi8-common-rpi linux-headers-6.1.0-rpi8-rpi-2712 linux-headers-6.1.0-rpi8-rpi-v8
  linux-image-6.1.0-rpi8-rpi-2712 linux-image-6.1.0-rpi8-rpi-v8 linux-kbuild-6.1
Utilice «sudo apt autoremove» para eliminarlos.
Se instalarán los siguientes paquetes NUEVOS:
  figlet
0 actualizados, 1 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
Se necesita descargar 134 kB de archivos.
Se utilizarán 750 kB de espacio de disco adicional después de esta operación.
Des:1 http://deb.debian.org/debian bookworm/main arm64 figlet arm64 2.2.5-3 [134 kB]
Descargados 134 kB en 0s (276 kB/s)
Seleccionando el paquete figlet previamente no seleccionado.
(Leyendo la base de datos ... 166762 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../figlet_2.2.5-3_arm64.deb ...
Desempaquetando figlet (2.2.5-3) ...
Configurando figlet (2.2.5-3) ...
update-alternatives: utilizando /usr/bin/figlet-figlet para proveer /usr/bin/figlet (figlet) en modo automático
Procesando disparadores para man-db (2.11.2-2) ...
              ____  _____ ____  _  ______ ___   ____  ____   ___
             |  _ \| ____/ ___|| |/ /  _ \_ _| |  _ \|  _ \ / _ \
             | | | |  _| \___ \| ' /| |_) | |  | |_) | |_) | | | |
             | |_| | |___ ___) | . \|  __/| |  |  __/|  _ <| |_| |
             |____/|_____|____/|_|\_\_|  |___| |_|   |_| \_\\___/

       ___ _   _ ____ _____  _    _     _        _  _____ ___ ___  _   _
      |_ _| \ | / ___|_   _|/ \  | |   | |      / \|_   _|_ _/ _ \| \ | |
       | ||  \| \___ \ | | / _ \ | |   | |     / _ \ | |  | | | | |  \| |
       | || |\  |___) || |/ ___ \| |___| |___ / ___ \| |  | | |_| | |\  |
      |___|_| \_|____/ |_/_/   \_\_____|_____/_/   \_\_| |___\___/|_| \_|

               ____ _____  _    ____ _____ ___ _   _  ____
              / ___|_   _|/ \  |  _ \_   _|_ _| \ | |/ ___|
              \___ \ | | / _ \ | |_) || |  | ||  \| | |  _
               ___) || |/ ___ \|  _ < | |  | || |\  | |_| |_ _ _
              |____/ |_/_/   \_\_| \_\|_| |___|_| \_|\____(_|_|_)

Could not able to download deskpi repo,will retry again. ... (warning).
Clonando en 'deskpi'...
remote: Enumerating objects: 703, done.
remote: Counting objects: 100% (276/276), done.
[... 58 lines cut ...]
    | | | |  _| \___ \| ' /| |_) | |  | |_) | |_) | | | | | |_ / _ \ |  \| |
    | |_| | |___ ___) | . \|  __/| |  |  __/|  _ <| |_| | |  _/ ___ \| |\  |
    |____/|_____|____/|_|\_\_|  |___| |_|   |_| \_\\___/  |_|/_/   \_\_| \_|

                     ____ ___  _   _ _____ ____   ___  _
                    / ___/ _ \| \ | |_   _|  _ \ / _ \| |
                   | |  | | | |  \| | | | | |_) | | | | |
                   | |__| |_| | |\  | | | |  _ <| |_| | |___
                    \____\___/|_| \_| |_| |_| \_\\___/|_____|

Please select speed level that you want:
It will always run at the speed level that you choosed.
---------------------------------------------------------------
1 - set fan speed level to 25%
2 - set fan speed level to 50%
3 - set fan speed level to 75%
4 - set fan speed level to 100%
5 - Turn off Fan
6 - Adjust the start speed level according to the temperature
7 - Cancel manual control and enable automatical fan control
---------------------------------------------------------------
Just input the number and press enter.
Your choice:

The installation was ok but the fan still doesnt work

yoyojacky, 2024-06-06:
> Again from beginning.
>
> 1. In this discussion, as the title suggests, we are writing about the not working installation script and driver for OS bookworm. I have raspberry  PI 8GB and bookworm OS [...]

DRIVERS folder is in the repository which you cloned from github.

yoyojacky, 2024-06-06:
> ould someboby write here the synthaxis with the exsact steps to proceed with the installation, please?

Hi ,do you 

> Hi, Effectively. I hace instaled it using
>
> ## git clone -b feature/bookworm https://github.com/deskpi-team/deskpi
> cd deskpi/installation/
> sudo ./install.sh
> `` Clonando en 'deskpi'... remote: [...]
when you execute this `deskpi-config` , you need to select one option by entry a number, such as `4` and observe the fan

kidekoop2016, 2024-06-07:
Hi,

> when you execute this deskpi-config , you need to select one option by entry a number, such as 4 and observe the fan

> 
Of course, I did it and it doesnt work

yoyojacky, 2024-06-17:
> Hi,
> 
> > when you execute this deskpi-config , you need to select one option by entry a number, such as 4 and observe the fan
> 
> > 
> 
> Of course, I did it and it doesnt work

Could you please check the executable file in /usr/bin/  which are copied from drivers folder?

[11 more replies]

closed 2024-07-05 by yoyojacky