DeskPi-Team/deskpi #158: Services are not getting activated

aravindjaimon, 2023-10-15:
The app is getting installed with the docs provided in https://wiki.deskpi.com/deskpipro/.
But after restart the fans are not getting persisted. When checking the status of the services, both services related to deskpi is on an activating state.

```
aj@nas:~ $ sudo systemctl status deskpi
● deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: activating (start) since Sun 2023-10-15 21:21:18 IST; 4min 33s ago
   Main PID: 3217 (pwmFanControl64)
      Tasks: 1 (limit: 8755)
        CPU: 106ms
     CGroup: /system.slice/deskpi.service
             └─3217 /usr/bin/pwmFanControl64 &
aj@nas:~ $ sudo systemctl status deskpi-cut-off-power.service
● deskpi-cut-off-power.service - DeskPi-cut-off-power service
     Loaded: loaded (/lib/systemd/system/deskpi-cut-off-power.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
aj@nas:~ $ sudo systemctl start deskpi-cut-off-power.service
^C
aj@nas:~ $ sudo systemctl status deskpi-cut-off-power.service
● deskpi-cut-off-power.service - DeskPi-cut-off-power service
     Loaded: loaded (/lib/systemd/system/deskpi-cut-off-power.service; enabled; vendor preset: enabled)
     Active: activating (start) since Sun 2023-10-15 21:26:27 IST; 15s ago
   Main PID: 3517 (safecutoffpower)
      Tasks: 1 (limit: 8755)
        CPU: 243ms
     CGroup: /system.slice/deskpi-cut-off-power.service
             └─3517 /usr/bin/safecutoffpower64
aj@nas:~ $

```

paulwratt, 2023-11-03:
did you try:
```
sudo systemctl restart deskpi-cut-off-power.service
```
I have just been play with disabling a buck of services, and `deskpi-cut-off-power.service` is installed & enabled, but was not in the `systemd-analyze blame` list

yoyojacky, 2024-05-13:
> did you try:
> 
> ```
> sudo systemctl restart deskpi-cut-off-power.service
> ```
> 
> I have just been play with disabling a buck of services, and `deskpi-cut-off-power.service` is installed & enabled, but was not in the `systemd-analyze blame` list.
> 
> Its likely that power service does not get started until a power-off state (halt, poweroff, reboot) is initiated by SystemD

Yes, that's right, once you initializing the shutdown sequnces, it will trigger the cut-off-power serivce to send "power_off" signal to /dev/ttyUSB0  which is listening by MCU on daughter board, and once the MCU get the "power_off" singal, it will cut off the power, and then, the blue light will turn red on power buttom.

closed 2024-05-13 by yoyojacky