DeskPi-Team/deskpi #159: Is it possible to use deskpi ssd as an external SSD for another computer (for imaging)

dmayle, 2023-10-27:
I'd like to re-image the SSD, and I'm wondering if it's possible to do so without pulling the drive from the deskpi case.

If I unplug the USB-A to USB-A adapter, can I use a USB-A to USB-A adapter/cable to plug the SSD as an external drive into another computer?  If so, will power for the SSD come via the cable?

Thanks!

paulwratt, 2023-11-03:
unplug USB DeskPi _connector_ from DeskPi and plug in a USBA-to-USBA-data-only cable, then  and give it a try, then report back please.

it _might_ work, I have an older 3.5 adaptor board on a RPi2/3 and it _does_ allow that, but the cable is _data only_ and requires the the hardware to already be powered

countertony, 2024-01-18:
@dmayle has this been attempted, and if so did it work? Enquiring minds etc...

dmayle, 2024-01-20:
By the time I got a response, I had already unplugged it and disassembled the case, so I have no idea

closed 2024-01-22 by dmayle