DeskPi-Team/deskpi #162: Deskpi drivers do not work - Ubuntu 23.10

Waggleworx, 2023-11-06:
I am on a stock installation of Ubuntu 23.10 (64-bit)

Installed by using the instructions in the wiki as follows.

For Ubuntu 64bit OS:
cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install-ubuntu-64.sh
sudo ./install-ubuntu-64.sh

This didn't work, which was already a bad sign.  I decided to dig into related issues.  [I followed the suggestions here](https://github.com/DeskPi-Team/deskpi/issues/146) without success, unfortunately.

Currently, deskpi.service is in a state where it will start, but is killed whenever deskpi-config is used.

systemctl status deskpi.service
● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-11-06 16:37:35 EST; 49min ago
   Main PID: 4008 (sudo)
      Tasks: 2 (limit: 8724)
     Memory: 1.3M
        CPU: 1.529s
     CGroup: /system.slice/deskpi.service
             ├─4008 sudo /usr/bin/pwmFanControl64
             └─4012 /usr/bin/pwmFanControl64

Nov 06 16:37:35 Dock systemd[1]: Started deskpi.service - DeskPi PWM Control Fan Service.
Nov 06 16:37:35 Dock sudo[4008]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl64
Nov 06 16:37:35 Dock sudo[4008]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)

sudo deskpi-config
[sudo] password for [user]: 
Welcome to Use DeskPi-Team's Product
Please select speed level that you want: 
It will always run at the speed level that you choosed.
---------------------------------------------------------------
1 - set fan speed level to 25%
2 - set fan speed level to 50%
3 - set fan speed level to 75%
4 - set fan speed level to 100%
5 - Turn off Fan
6 - Adjust the start speed level according to the temperature
7 - Cancel manual control and enable automatical fan control
---------------------------------------------------------------
Just input the number and press enter.
Your choice:4
You'tve select 100% speed level
Fan speed level has been change to 100%

systemctl status deskpi.service
○ deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; preset: enabled)
     Active: inactive (dead) since Mon 2023-11-06 17:27:37 EST; 3s ago
   Duration: 50min 2.653s
    Process: 4008 ExecStart=sudo /usr/bin/pwmFanControl64 (code=killed, signal=TERM)
   Main PID: 4008 (code=killed, signal=TERM)
        CPU: 1.539s

Nov 06 16:37:35 Dock systemd[1]: Started deskpi.service - DeskPi PWM Control Fan Service.
Nov 06 16:37:35 Dock sudo[4008]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl64
Nov 06 16:37:35 Dock sudo[4008]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Nov 06 17:27:37 Dock sudo[4008]: pam_unix(sudo:session): session closed for user root
Nov 06 17:27:37 Dock systemd[1]: Stopping deskpi.service - DeskPi PWM Control Fan Service...
Nov 06 17:27:37 Dock systemd[1]: deskpi.service: Deactivated successfully.
Nov 06 17:27:37 Dock systemd[1]: Stopped deskpi.service - DeskPi PWM Control Fan Service.
Nov 06 17:27:37 Dock systemd[1]: deskpi.service: Consumed 1.539s CPU time.

Please advise on how to make this remotely functional.

If I'm honest, the typos spread throughout documentation, and deskpi-config do not inspire confidence.

Waggleworx, 2023-11-06:
Even after removing all services and doing a plain reinstallation with the ubuntu-64 script, my services are unusable:

○ deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; preset: enabled)
     Active: inactive (dead) since Mon 2023-11-06 17:48:53 EST; 40s ago
   Duration: 1min 29.647s
    Process: 1576 ExecStart=sudo /usr/bin/pwmFanControl (code=killed, signal=TERM)
   Main PID: 1576 (code=killed, signal=TERM)
        CPU: 96ms

Nov 06 17:47:24 Dock systemd[1]: Started deskpi.service - DeskPi PWM Control Fan Service.
Nov 06 17:47:24 Dock sudo[1576]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
Nov 06 17:47:24 Dock sudo[1576]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Nov 06 17:48:53 Dock sudo[1576]: pam_unix(sudo:session): session closed for user root
Nov 06 17:48:53 Dock systemd[1]: Stopping deskpi.service - DeskPi PWM Control Fan Service...
Nov 06 17:48:53 Dock systemd[1]: deskpi.service: Deactivated successfully.
Nov 06 17:48:53 Dock systemd[1]: Stopped deskpi.service - DeskPi PWM Control Fan Service.
~

yoyojacky, 2023-11-20:
> [Waggleworx](/Waggleworx)

Hi waggleworx, could you please send me your ubuntu verson  and release infor?

yoyojacky, 2023-11-23:
it may cause by the script permission, let me check new version of the Ubuntu OS and give you a feedback, Could you please tell me where the typo is?

yoyojacky, 2024-05-20:
> I am on a stock installation of Ubuntu 23.10 (64-bit)
>
> Installed by using the instructions in the wiki as follows.
>
> For Ubuntu 64bit OS: cd ~ git clone https://github.com/DeskPi-Team/deskpi.git cd [...]

Hi 

> I am on a stock installation of Ubuntu 23.10 (64-bit)
>
> Installed by using the instructions in the wiki as follows.
>
> For Ubuntu 64bit OS: cd ~ git clone https://github.com/DeskPi-Team/deskpi.git cd [...]

Hi @Waggleworx , 
Thanks for your tips  and I am work on it and find that once you execute `deskpi-config` command, it will take over the control permision from systemd service, and it will stop the service and occupy the current serial port, I am work on how to resume the service after you execute the `deskpi-config` command, there is a way to resume the service is to execute `sudo systemctl restart deskpi.service`.   
If you have any problem with it ,please discuss the issue in this threads.

yoyojacky, 2024-07-04:
Hi waggleworx, it seems like the ubuntu 24.04 LTS will not support deskpi pro due to the function module : dwc2 does not work properly, and it will not take effect even though  it has been added `dtoverlay=dwc2,dr_mode=host` and reboot the pi, it will not recognized the MCU's serial port : `/dev/ttyUSB0` and it will not enable front usb ports as well. 
we have to remove this OS from our supported list due to this OS's BUG.

closed 2024-07-04 by yoyojacky