DeskPi-Team/deskpi #170: [FR] LineageOS 20 (Android 13) support for DeskPi Pro v3

azumimuo, 2023-12-03:
Hello..

Is there any chance to have some support for Android build?

Currently, I'm successfully enable dual front USB port by adding "dtoverlay=dwc2,dr_mode=host" in /boot/config.txt and successfully boot M.2 SATA SSD from USB 3.0 port.

I'm hoping that someone could share some android script because I do not have basic programming and I do know how to control or start the pwm fan and also add safe-cut-off power switch features.

Thanks
Azumimuo

yoyojacky, 2024-01-30:
sorry, it does not support android OS.

closed 2024-01-30 by yoyojacky

azumimuo, 2024-05-12:
Hello again..

Since no support for Android, would you share the location of constant 3V/5V output point on-board (either than usb pin or gpio pin) power for cooling fan?

Thanks
-Azumimuo-

yoyojacky, 2024-05-13:
Hi Azumimuo, 
You mean here ？
[image]

yoyojacky, 2024-05-13:
or you can just use the GPIO Pin which is spread out by using FFC cable to the daughter board of deskPi pro
[image]

azumimuo, 2024-05-13:
>  Hi Azumimuo, 
> You mean here ？
> [image]
> 

it's only activated only by working script (currently, no android supported script)

azumimuo, 2024-05-13:
> or you can just use the GPIO Pin which is spread out by using FFC cable to the daughter board of deskPi pro
> [image]
> 

[image]

already done that. but don't like it.

yoyojacky, 2024-05-15:
use GPIO instead of it ?

azumimuo, 2024-05-16:
> use GPIO instead of it ? 
> 

yes, not connected to the internal fan connector. there's no android supported pwm fan script for DeskPi Pro v3.

yoyojacky, 2024-06-06:
Oh ,  I see. yes ,it is hard to adding android support pwm fan script due to the variable version of android system. and it may need to recompile kernel or something I do not know.

azumimuo, 2024-06-22:
hello Deskpi technical team & yoyojacky, are this unsoldered poin named "USB3" can give constant 5 volt?

[image]

yoyojacky, 2024-07-06:
[image]
why don't you use this Fan cable for 5V fan output ? 
or you want to use other 5v device ?
you can use GPIO on back -> 
[image]

yoyojacky, 2024-07-06:
[image]
 Here is the pin out of USB3 soldering point. 
it can provide constant 5V and GND as figure showing.