DeskPi-Team/deskpi #173: Front USB and Fan not working on RaspberryPi 4 (Buster or Bullseye / 32 or 64 bits)

therestoftheday, 2024-01-16:
Hello,

I have tried everything to make front USB and Fan work but none of them are working.

```
uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

```
cat /proc/version
Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
```

```
cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.5
```

I have tried Raspbian Buster 32-bit version (the one shown here : https://github.com/DeskPi-Team/deskpi/), with install.sh script, Debian Bullseye 32-bit version with install.sh script, and Debian Bulleseye 64-bit version with install-raspbian-64.sh script.
Nothing works.

--------

**For the fan** : deskpi.service seems to work, but when I try to make it work at 100%...

```
Just input the number and press enter.
Your choice:4
You'tve select 100% speed level
Fan speed level has been change to 100%
```
then ...
```
deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset>
     Active: failed (Result: signal) since Tue 2024-01-16 16:38:33 CET; 7s ago
    Process: 642 ExecStart=/usr/bin/pwmFanControl64 & (code=killed, signal=TERM)
   Main PID: 642 (code=killed, signal=TERM)
        CPU: 114ms

janv. 16 16:30:31 Spectra systemd[1]: Starting DeskPi Fan Control Service...
janv. 16 16:38:33 Spectra systemd[1]: deskpi.service: Main process exited, code>
janv. 16 16:38:33 Spectra systemd[1]: deskpi.service: Failed with result 'signa>
janv. 16 16:38:33 Spectra systemd[1]: Stopped DeskPi Fan Control Service.
lines 1-11/11 (END)...skipping...
● deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Tue 2024-01-16 16:38:33 CET; 7s ago
    Process: 642 ExecStart=/usr/bin/pwmFanControl64 & (code=killed, signal=TERM)
   Main PID: 642 (code=killed, signal=TERM)
        CPU: 114ms
```
--------
NB : When I make a long push on the power button, the fan works. And that's it. A small press stops it.

------------------------------

**For the front USB** : 
The ' dtoverlay=dwc2,dr_mode=host' stuff is correctly added to /boot/config.txt
But after (multiple) reboots, the front USB ports have never worked (i tested it by plunging a keyboard and a mouse on them).

--------

I have wandered through the multiple issues and made different tries, but I am really stuck. At start I thought the 64-bit version Raspbian and the 64-bit version of the deskpiscript were incompatible, but even 32-bit Raspbian version and 32-bit version of the script are not working 'together'.
I spent the day trying to make it work, with no success at all.
Thanks for your help.

yoyojacky, 2024-01-30:
For front USB: 
Did you try to reassemble Raspberry Pi and the daughter board and make sure the USB-C port connect to the daughter board properly ?

ani2life, 2024-02-27:
```bash
$ uname -a
Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
```

I encountered the same issue.

Upon checking the `/boot/config.txt` file, I found the following contents:

```
$ cat /boot/config.txt
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/config.txt
```

I added `dtoverlay=dwc2,dr_mode=host` to the end of the `/boot/firmware/config.txt` file, and after rebooting, the front USB ports and the fan started working.

I used the `install-ubuntu-64.sh` script for the deskpi installation.

ani2life, 2024-02-27:
I found that modifying the `/boot/config.txt` file on `Raspberry Pi OS Lite (64-bit)` I installed was ineffective. The correct solution was to modify the `/boot/firmware/config.txt` file.

yoyojacky, 2024-05-09:
please try new script in branch : feature/bookworm 's installation folder.

yoyojacky, 2024-05-09:
[image]

closed 2024-05-11 by yoyojacky