DeskPi-Team/deskpi #175: Unable to install driver on DietPi 64 bit Bullseye

Geppetto24, 2024-06-01:
I'm attempting to install the DeskPi Pro V2 driver to a current DietPi 64 bit Bullseye installation (Raspberry Pi 4B with 4 GB mem, SD card), using the instructions from  https://github.com/DeskPi-Team/deskpi, but install.sh appears to be missing from the repository.  Does anyone know of a workaround solution?  Thank you.

yoyojacky, 2024-06-25:
Hi Geppetto24, 
I have already fixed the bug, please download latest repository and try again.

closed 2024-06-25 by yoyojacky

Geppetto24, 2024-06-29:
Thank you for the update.   I installed as directed on a fresh dietpi 64bit bookworm installation with amiberry.  The driver did install, however, the fan does not turn on nor does the power button turn red upon power off. 
Does the driver only work with bullseye, or is amiberry causing a conflict perhaps?  
Thank you. 

Closed #175 as completed.

yoyojacky, 2024-07-01:
Could you please try to check if there is a `/dev/ttyUSB0` device has been recognized by your DietPi OS?  Beacuse the fan is controlled by the MCU on daughter board and this device is created by adding `dtoverlay=dwc2,dr_mode=host` parameter in `/boo/firmware/config.txt` file, and if your DietPi OS supports the dtoverlay, it will works properly, and at default, if your CPU's temperature has not been reached to the temperature that you specified in `/etc/deskpi.conf` file ,it will not turn on the fan, please  try to execute `deskpi-config` --> `4` to try if the fan can be triggered.  and the safe_cut_off_power service  will be triggered once you shutting down the system, for example,  when you execute `sudo init 0` or `sudo shutdown -h now` , or just click `shutdown system` icon in the menu. and  deskpi pro v3 can support most Linux system, but it needs to install driver to control the fan and the power supply indicator. 
There are some tips for you:
1.  Make sure the OS can support `dtoverlay=dwc2,dr_mode=host` to genetrate `/dev/ttyUSB0` device in your OS.
2. Make sure your account has permission to operate the `/dev/ttyUSB0` device 
3. Install the deskpi pro driver by executing the shell script on this repository. 
4. Make sure the system file has been created successfully in `/etc/systemd/system/deskpi.service` and it will copy the binary executable file to your `/usr/bin/pwmFanControl64`  from repository's `installation/drivers/c/` folder. and there are several source file in that folder ,you can compile it by yourself too. 
5. Make sure the system file `/etc/systemd/system/deskpi.saft_cut_off_power.service`  has been created successfully, and it will copy a binary file from the same folder to `pwmFanControl64` file. 
6.  After installation ,please reboot your Raspberry Pi and open a terminal and typing `deskpi-config` --> `4` to check if the fan can running in full speed mode, and input `5` to stop the fan, and then you can input `6` to create a file in `/etc/deskpi.conf`, it will require you input the threshold of temperature and the speed of fan from 0-100, for example :
45   # temperature 
75   # fan speed in pwm signal 
50   # temperatrue 
100  # fan speed in pwm signal 
55    # temperatrue 
100  # fan speed in pwm signal 
60     # temperatrue 
100 # fan speed in pwm signal 
7. Do remember restart deskpi.service after using `deskpi-config` command 
```bash
sudo systemctl restart deskpi.service`
```
Hope it will help you out. have a nice day!

Geppetto24, 2024-07-06:
Well the first problem I see is that the installation script is looking for the confix.txt file in the incorrect location. Dietpi bookworm 64 bit does not have it located in /boot/firmware/. It is still /boot/config.txt as in previous versions (unlike raspberry pi os).  I edited the install script to use the correct path, and it did install.  I do see that ttyUSB0 is available, however, the fan still does not run nor does the power button script.  
The product is far too complicated for the average raspberry pi user.

yoyojacky, 2024-07-20:
> Make sure the system file has been created successfully in /etc/systemd/system/deskpi.service and it will copy the binary executable file to your /usr/bin/pwmFanControl64 from repository's [...]

1. Could you please check the /usr/bin/ location and check if the pwmFanControl64 exists?