DeskPi-Team/deskpi #180: Various issues

magorium, 2024-07-06:
64-bit raspbian OS (or whatever they call it these days), deskpi pro V2

1. The install script attempts to copy the c utils from the /tmp directory which does not exist nor does the readme instructions mention anything related to /tmp. Please fix one or the other.
2. the deskpi-config script uses figlet on line 46 which isn't installed by recent 64-bit default desktop image. That results in an unknown command error every time you run the script. Please refrain from using 3th party apps/tools, install it for the user or make it optional in the script.
3. I seem unable to control the fans. Probably related to ttyusb0 issue because I have a mouse a keyboard and a SSD connected at boot. Please fix this.
4. Manually installing the c utils and when attempting to shutdown from the desktop the system hangs on the plymouth shutdown service. I did not have this issue before running the install script. Please advise.
5. the USB front ports do not seem to be usable/working. Please advise.
6. Don't know if it is related to the deskpi itself but before running the script the desktop OS constantly complained about low voltage errors using the deskpi delivered power supply. After running the script (but it might also the power down and re-start) this issue seem to have resolved itself (though I am not sure if the issue /is/ actually resolved. Please advise

regards,

yoyojacky, 2024-07-06:
Hi, magorium, 
Did  you use the latest installation script in https://github.com/DeskPi-Team/deskpi/blob/master/installation/RaspberryPiOS/64bit/install-raspios-64bit.sh ?
The installation scripts has been updated for a while and the figlet has been removed from the mainline, you can just use the latest one but not the branch. 
and the script will not download the repository to /tmp again, it has been removed from script already. 

2. If you don't like the `figlet` style, please modify `/usr/bin/deskpi-config` file, comment out the figlet part, and instead it with echo will be ok. but i suggest  you re-download the repository again and use the latest script to install the driver. 
3.  Please check if the `dtoverlay=dwc2,dr_mode=host`  has been added into `/boot/firmware/config.txt`  file and do remember reboot your Raspberry Pi 4B to take effect. 
4. It is not hangs on the plymouth shutdown service, the shutdown service will trigger safe_cut_off_power.service once you attempting to shutdown raspberry pi, it will send `poweroff` signal to `/dev/ttyUSB0` device, so that the MCU on deskpi pro daughter board will know you are trying to shutdown the Raspberry Pi, and it will wait for 10 seconds to execute the cutting off power operation, once done, it will cut off the power and turn the power button's LED from blue to red. 
5. If the Front USB port dose not work properly, please try to re-assemble the case, check the /boot/firmware/config.txt file's configuration: `dtoverlay=dwc2,dr_mode=host` . and try to reboot raspberry pi without SSD connected.  
6. If you constantly complained about low voltage errors during using deskpi pro v2, please contact our sales for replacement. 
7.  If you have any problem on using this product, please kindly let me know, and have a nice day!

Best regards

magorium, 2024-07-06:
Hello yoyojacky,

Thank you for the quick reply.

> Did you use the latest installation script in https://github.com/DeskPi-Team/deskpi/blob/master/installation/RaspberryPiOS/64bit/install-raspios-64bit.sh ?

Yes, I did use the latest installation script as mentioned. That was the reason for me to address it :-)

1.
Using trunk (= main/master):

```
$ mkdir work&&cd work
$ git clone https://github.com/DeskPi-Team/deskpi.git
$ cd deskpi
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
$ grep "/tmp/" installation/RaspberryPiOS/64bit/install-raspios-64bit.sh
if [ -d /tmp/deskpi/ ]; then
        cp -Rvf /tmp/deskpi/installation/drivers/c/pwmFanControl64 /usr/bin/pwmFanControl64
        cp -Rvf /tmp/deskpi/installation/drivers/c/safeCutOffPower64 /usr/bin/safeCutOffPower64
        cp -Rvf /tmp/deskpi/installation/deskpi-config  /usr/bin/deskpi-config
```
Nowhere does the readme state that a different branch needs to be checked out before running the script Therefor I assume that the above output is as intended. Thus I am sorry to have to disagree with your statement that the /tmp/ lines have been removed from the script.

If you have a different branch where this situation is corrected then these corrections needs to be merged back to trunk (= main/master) or the name and checkout of the correct branch needs to be mentioned in the readme.

2. Thank you for the suggestion to remove the use of figlet in the deskpi-config script but it is imho besides the point.
Nowhere (in the documentation) is it mentioned that there is a 3th party dependency. Therefor I suggest to mention it in the documentation (installation instructions) or the script must be able to work around it (and not depend on the presence of this 3th party util and only use it when present).

3. My boot's config.txt reads:

~~~
$ cat /boot/firmware/config.txt 
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
~~~

As can be seen the dtoverlay is present at the section named [cm5]

4. Thank you for explaining the situation where I (wrongfully) assumed that
the plymouth shutdown service is 'hanging'. I will have to check that and report back
if it was indeed my impatience. For now I have to assume that if it checks hardcoded /dev/ttyUSB0 device that
this might cause issues as mentioned before (acessing the wrong USB device because of system ordering based on what devices are plugged into the pi)

5. Thank you for the suggestion regarding the front USB that appears to me as not working.
I will reassemble the case but booting it without SSD will take me some time as I do not have
a spare SD card (so need to order one). I will report back on that later.

6. regarding the low voltage message from the desktop version of raspbian I stipulate (again) that these messages disappeared as soon as the installation script finished installing (and I manually copied the tools).

7. Thank you very much for your kind words and prompt reply. Despite the issues I am running into
I really like the case. I ordered the Deskpi Pro just before the shortage of Raspberry Pi's and therefor had to wait almost 2 years before being able to actually assemble it (which is now). You have a nice day yourself as well.

regards,

Meusique, 2024-07-08:
Hi there:

I can confirm that the current version of the script no longer uses figlet. However, deskpi-config does -- it throws an error if you run it without figlet installed. It still runs, but it won't display the fancy ASCII logo.

I'm having similar issues getting the fan to work, though. I'm using a V2 DeskPi Pro, with a Pi 4 running a fresh installation of the latest version of RaspberryPiOS Bookworm. My config.txt includes `dtoverlay=dwc2,dr_mode=host`. Everything in the case is seated snugly; the power button works as expected and the microSD slot works just fine.

As stated above, the install script attempts to copy the drivers and deskpi-config from `/tmp/deskpi/`, but that's not where they are if you follow the instructions on the man page -- everything is in `~/deskpi/` (because that's where the instructions tell you to clone the git repository to).

From what I can tell, this is a leftover from an older version of the install script. It used to start by cloning the repo to `/tmp/deskpi/`. The current version doesn't do that, so when it gets to the part where it tries to copy the drivers and deskpi-config to `/usr/bin/`, nothing happens. So, the script needs to be updated to either start by cloning the repo to `/tmp/deskpi/` again, like it used to, or look for the files in `~/deskpi/` instead. The first option is more foolproof, although it does mean cloning the repo a second time -- the second option is lighter weight, but means the script breaks if the repo hasn't been cloned to `~/deskpi/`.

But... even if I manually copy the drivers and deskpi-config to the correct location in `/usr/bin/`, I still don't seem to be able to control the fan. The settings in deskpi-config do nothing; even setting it to 100% does nothing.

Also, the front USB ports only sort of work -- they supply power (I can use them to power a USB light, or charge a phone), but the system doesn't recognize any devices I plug in -- devices that work fine when I plug them into the back USB ports. This is a minor issue -- I don't plan to use them for anything other than power. But maybe it helps shed some light on what's going wrong.

I'm not a programmer, but here's my attempt at trying to debug this. When I try to run pwmFanControl64 manually, I get the following error:

```
`Can not open /dev/ttyUSB0 serial port ErrorCode: No such file or directory
Please check the /boot/config.txt file and add dtoverlay=dwc2, dr_mode=host and reboot RPi 
Please check serial port over OTG
---Serial Port Can not detected---`
```

When I try to run safeCutOffPower manually, it throws this error in a loop forever, filling up my terminal over and over again with:

```
Can not access /dev/ttyUSB0, please check it out.
serial port can not be accessed
```

So clearly something is wrong with the serial port interface. Sure enough, if I cd to `/dev/` and ls, there's no `ttyUSB0` listed.

Any idea what the problem is here, or how to fix it? I haven't changed anything from a fresh install of RaspberryPiOS other than setting my username, password and local via config.txt, running apt update and upgrade, and installing git to clone the repo.

Thanks!

magorium, 2024-07-09:
@Meusique:
Indeed, your description is the more elaborated version of what I wrote :-)

> Any idea what the problem is here, or how to fix it?

Yes, it is begginning to daunt me what the issue actually is. Unfortunately, I do not have a step by step solution  (yet, and in case possible).

Right now I'm cutting down trees in order to try and attempt to have a glimpse at the forest.

@yoyojacky:
regarding point 5.

I disassembled and reassembled my deskpi pro v2 and while it was interesting enough to do so it was a utter waste of time and time that I'll never be able to get back.

It would have been more productive would you have pointed to the closed issue (#157) that exactly describes my (and other people's) situation.

When a topic is closed then it is assumed that the issue has been dealt with, and obviously it has not (which is fine on itself but just acknowledge it as it saves both you and me a lot of time ;-) ).

BTW, I checked all existing branches and they all use the hardcoded /tmp/ folder and to make maters worse trunk uses that for at least 3 years so trunk script could never have worked for anyone (except for those that are able to manually correct things).

So I now need to figure out what the minimal overlay version is that is able to work for a pi4 icw determining which eeprom version plays nice with that and then address the existing issues in trunk.

The reason that I perhaps sound harsh is because I did not expected to have to get this much involved in something that seems trivial. So, please forgive me.

regards,

magorium, 2024-07-09:
@Meusique:
Indeed, your description is the more elaborated version of what I wrote :-)

> Any idea what the problem is here, or how to fix it?

It is beginning to daunt me :-)

Can you open up a terminal and issue the command:
```
cat /boot/firmware/config.txt
```
And paste the complete output of that file for me/us ?

@yoyojacky
regarding point 5:
Disassembling and re-assembling, while interesting enough on its own, did not help. Was it so hard to let me know that these sections actually meant something ? ;-p

Meusique, 2024-07-09:
Here's my config.txt output:

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
enable_uart=1
```

Meusique, 2024-07-09:
I figured it out: in my config file, `dtoverlay=dwc2,dr_mode=host` is under `[cm5]`, meaning it's only active when running on a CM5 unit, i.e. a Raspberry Pi Compute Module 5, which as far as I can tell, doesn't actually exist. Moving `dtoverlay=dwc2,dr_mode=host` up to the top of the config file, so it's applied in all cases, means that `/dev/ttyUSB0` now shows up and fan control now works. Hooray!

Seems like the fix here would be for the install script to explicitly check not only whether `dtoverlay=dwc2,dr_mode=host` is in config.txt, but whether it's under `[cm5]`, as it seems to be by default in RaspberryPiOS right now (because I didn't modify config.txt at all prior to attempting to install the DeskPi software).

(That and fixing the other current issue of the script, where it attempts to copy files from `/tmp/deskpi/` without either directing the user to clone the repo there first, or cloning it there itself.)

Now I just need to figure out the noisy fan issue... (https://github.com/DeskPi-Team/deskpi/issues/25)

Meusique, 2024-07-09:
@magorium

> [cm5]
> dtoverlay=dwc2,dr_mode=host
> ```

Looks like you've got the same issue I had in your config.txt. Try doing what I did and see if it works for you.

magorium, 2024-07-10:
Thank you for the suggestion @Meusique.

I already figured that out when I wrote my previous reply (hence why I asked to post your config in order to see if you had the same issue)

The technical term for that is named conditional filtering and the documentation can be found [here](https://www.raspberrypi.com/documentation/computers/config_txt.html#conditional-filters).

I fixed mine by adding a pi4 section and adding the overlay line manually.

Sorry for the late(r) reply but shortly after I got things to work I got haunted by the undervolted notifications again and shortly after that the power supply started making a hissing high pitched noise (changing its pitch higher and higher as time went on) and when I hasted myself to get it unplugged the power supply shorted out.

So, I have to assume that my power supply that came with the Deskpi Pro V2 has been faulty from the start especially since using an official raspberry pi power adapter did not result in the same symptoms.

You are correct that the (current) install script does not seem to properly recognizes the conditional filtering and there for makes the wrong assumption(s).

I am sorry to say that I can't really help with the noisy fan. Mine does hum a little at lower revolutions but at 100% the hum is nearly gone and the fan sounds like a normal fan. I can only suggest to make sure that the fan area is clear from any obstructions and try to see if the fan or fan blades touches anything. tbh, I do not even know if it is possible to remove the fan from the ice-cooler (in order to bend some blades and/or the fan or be able to replace it).

As it stands now and afaik at this point, there are 3 issues left:
1. install script does not correctly identifies the presence of the overlay and therfor does not add it to the config.txt file
2. install script is unable to copy the utilities due to wrong use of hardcoded /tmp directory
3. deskpi-config script uses figlet even when not installed on users system nor does it install figlet for the user or does the readme states that figlet is a 3th party dependency.

yoyojacky, 2024-07-20:
> 你好，yoyojacky，
> 
> 感谢你能这么快回复。
> 
> > [您是否使用了https://github.com/DeskPi-Team/deskpi/blob/master/installation/RaspberryPiOS/64bit/install-raspios-64bit.sh](https://github.com/DeskPi-Team/deskpi/blob/master/installation/RaspberryPiOS/64bit/install-raspios-64bit.sh)中的最新安装脚本？
> 
> 是的，我确实使用了上述最新的安装脚本。这就是我解决这个问题的原因 :-)
> 
> 
> 使用主干 (= 主/主控)：
> 
> ```
> $ mkdir work&&cd work
> $ git clone https://github.com/DeskPi-Team/deskpi.git
> $ cd deskpi
> $ git status
> On branch master
> Your branch is up to date with 'origin/master'.
> 
> nothing to commit, working tree clean
> $ grep "/tmp/" installation/RaspberryPiOS/64bit/install-raspios-64bit.sh
> if [ -d /tmp/deskpi/ ]; then
>         cp -Rvf /tmp/deskpi/installation/drivers/c/pwmFanControl64 /usr/bin/pwmFanControl64
>         cp -Rvf /tmp/deskpi/installation/drivers/c/safeCutOffPower64 /usr/bin/safeCutOffPower64
>         cp -Rvf /tmp/deskpi/installation/deskpi-config  /usr/bin/deskpi-config
> ```
> 
> 自述文件中没有任何地方指出在运行脚本之前需要签出不同的分支，因此我认为上述输出符合预期。因此，我很抱歉不同意您关于已从脚本中删除 /tmp/ 行的说法。
> 
> 如果您有不同的分支可以纠正这种情况，那么这些更正需要合并回主干（= main/master）或者需要在自述文件中提及正确分支的名称和签出。
> 
> 2. 感谢您建议在 deskpi-config 脚本中删除 figlet 的使用，但在我看来这与主题无关。
>    文档中没有任何地方提到存在第三方依赖项。因此，我建议在文档（安装说明）中提及它，否则脚本必须能够绕过它（并且不依赖于此第三方实用程序的存在，并且仅在存在时使用它）。
> 3. 我的启动的 config.txt 内容如下：
> 
> ```
> $ cat /boot/firmware/config.txt 
> # For more options and information see
> # http://rptl.io/configtxt
> # Some settings may impact device functionality. See link above for details
> 
> # Uncomment some or all of these to enable the optional hardware interfaces
> #dtparam=i2c_arm=on
> #dtparam=i2s=on
> #dtparam=spi=on
> 
> # Enable audio (loads snd_bcm2835)
> dtparam=audio=on
> 
> # Additional overlays and parameters are documented
> # /boot/firmware/overlays/README
> 
> # Automatically load overlays for detected cameras
> camera_auto_detect=1
> 
> # Automatically load overlays for detected DSI displays
> display_auto_detect=1
> 
> # Automatically load initramfs files, if found
> auto_initramfs=1
> 
> # Enable DRM VC4 V3D driver
> dtoverlay=vc4-kms-v3d
> max_framebuffers=2
> 
> # Don't have the firmware create an initial video= setting in cmdline.txt.
> # Use the kernel's default instead.
> disable_fw_kms_setup=1
> 
> # Run in 64-bit mode
> arm_64bit=1
> 
> # Disable compensation for displays with overscan
> disable_overscan=1
> 
> # Run as fast as firmware / board allows
> arm_boost=1
> 
> [cm4]
> # Enable host mode on the 2711 built-in XHCI USB controller.
> # This line should be removed if the legacy DWC2 controller is required
> # (e.g. for USB device mode) or if USB support is not required.
> otg_mode=1
> 
> [cm5]
> dtoverlay=dwc2,dr_mode=host
> 
> [all]
> ```
> 
> 可以看出，dtoverlay 位于名为 [cm5] 的部分
> 
> 4. 感谢您解释我（错误地）认为 plymouth 关机服务“挂起”的情况
>    。我必须检查一下，
>    如果确实是我的不耐烦，我会报告。现在我不得不假设，如果它检查硬编码的 /dev/ttyUSB0 设备，
>    这可能会导致前面提到的问题（由于系统根据插入 pi 的设备进行排序，因此访问错误的 USB 设备）
> 5. 感谢您就前置 USB 提出的建议，在我看来，它似乎无法正常工作。
>    我会重新组装机箱，但由于我没有
>    备用 SD 卡（因此需要订购一张），因此在没有 SSD 的情况下启动它会花费一些时间。我稍后会报告此事。
> 6. 关于来自 raspbian 桌面版的低电压消息，我（再次）规定，这些消息在安装脚本完成安装后立即消失（我手动复制了工具）。
> 7. 非常感谢您的善意评价和及时回复。尽管我遇到了一些问题，
>    但我真的很喜欢这个外壳。我在 Raspberry Pi 短缺之前订购了 Deskpi Pro，因此不得不等待近 2 年才能够真正组装它（现在已经组装好了）。祝您也有美好的一天。
> 
> 问候，

oh, magorium, that's my bad, seems like the brench is a little bit mess due to my mistake.  I will double check it again an fix the bug.

yoyojacky, 2024-07-20:
> ttyUSB0

Oh, I forget to remove the figlet on `deskpi-config` scripts, it will be fixed at next update. 
about the serial port can not be recognized, i think it is may connection problem, I have never met this before, because I have test a lot of Raspberry Pi on this deskpi pro case, but it never happened ,do you have another raspberry pi 4b you can test it on this caes?

yoyojacky, 2024-07-20:
> @magorium
> 
> > ```
> > [cm5]
> > dtoverlay=dwc2,dr_mode=host
> > ```
> 
> Looks like you've got the same issue I had in your config.txt. Try doing what I did and see if it works for you.

please move this parameter to :
```bash
[all]
dtoverlay=dwc2,dr_mode=host 
```
and do remember reboot your raspberry pi to take effect.

noanight, 2024-07-20:
install-raspios-64bit.sh seems to be fixed, but install-ubuntu-64.sh has still the same erroneous code.

```shell

# missing cp...

# copy pre-compiled binary file to /usr/bin/ folder
if [ -d /tmp/deskpi/ ]; then
        cp -Rvf /tmp/deskpi/installation/drivers/c/pwmFanControl64 /usr/bin/pwmFanControl64
        cp -Rvf /tmp/deskpi/installation/drivers/c/safeCutOffPower64 /usr/bin/safeCutOffPower64
        cp -Rvf /tmp/deskpi/installation/deskpi-config  /usr/bin/deskpi-config
        chmod +x /usr/bin/pwmFanControl64
        chmod +x /usr/bin/safeCutOffPower64
        chmod +x /usr/bin/deskpi-config
fi
```

yoyojacky, 2024-07-23:
> install-raspios-64bit.sh seems to be partially fixed, but install-ubuntu-64.sh has still the same missing git clone to /tmp/deskpi code.
> 
> ```shell
> # missing cp...
> 
> # copy pre-compiled binary file to /usr/bin/ folder
> if [ -d /tmp/deskpi/ ]; then
>         cp -Rvf /tmp/deskpi/installation/drivers/c/pwmFanControl64 /usr/bin/pwmFanControl64
>         cp -Rvf /tmp/deskpi/installation/drivers/c/safeCutOffPower64 /usr/bin/safeCutOffPower64
>         cp -Rvf /tmp/deskpi/installation/deskpi-config  /usr/bin/deskpi-config
>         chmod +x /usr/bin/pwmFanControl64
>         chmod +x /usr/bin/safeCutOffPower64
>         chmod +x /usr/bin/deskpi-config
> fi
> ```
> 
> install-raspios-64bit.sh and contain:
> 
> ```shell
> # check if dwc2 dtoverlay has been enabled.
> checkResult=`grep dwc2 /boot/firmware/config.txt`
> if [ $? -ne 0 ];  then
> ....
> ```
> 
> the config.txt for bookworm contains...
> 
> ```ini
> [cm5]
> dtoverlay=dwc2,dr_mode=host
> ...
> ```
> 
> the code above skips adding dtoverlay=dwc2,dr_mode=host to the [all] section which should be
> 
> ```
> [all]
> dtoverlay=dwc2,dr_mode=host 
> ```
> 
> maybe something like this could do the job?
> 
> ```shell
> CONFIG_FILE="boot/firmware/config.txt"
> SECTION="[all]"
> LINE_TO_CHECK="dtoverlay=dwc2,dr_mode=host"
> 
> # Check if the [all] section exists in the file
> if grep -q "\[all\]" "$CONFIG_FILE"; then
>     # Check if the LINE_TO_CHECK exists within the first 10 lines after [all] section
>     if grep -A 10 "\[all\]" "$CONFIG_FILE" | grep -q "$LINE_TO_CHECK"; then
>         echo "Line already exists in the [all] section."
>     else
>         echo "Adding line to the [all] section."
>         # Add the LINE_TO_CHECK below the [all] section
>         sed -i "/\[all\]/a$LINE_TO_CHECK" "$CONFIG_FILE"
>     fi
> else
>     echo "Section [all] not found in the config file."
> fi
> ```

yes, it will be fixed at once~

yoyojacky, 2024-07-23:
> install-raspios-64bit.sh seems to be partially fixed, but install-ubuntu-64.sh has still the same missing git clone to /tmp/deskpi code.
>
> ```shell
> # missing cp...
>
> # copy pre-compiled binary file to [...]

Hi bro, 
The script has been updated. thanks for your tips, have a nice day!

noanight, 2024-07-23:
hey yoyojack,
I’ve been using the GeeekPi DeskPi Pro V3 since 2023. It would be helpful if the software will be tested more thoroughly. The setup for raspios64 Bullseye was bumpy too.
Have a nice day!

yoyojacky, 2024-07-23:
Hi noanight, 
Thanks for your tips, and if you want to use bullseye, you can try our pre-installed image : 
[Pre-installed-image-bullseye-64bit-for-deskpi-pro](https://drive.google.com/file/d/1fKJ6ZpMaEAjIvEa72R3Y0AAQbiwLF76c/view?usp=drive_link)

Well, when we tested, there were some differences in the environments of different systems. Now they require bookworm and bullseye to make pre-installed images. 
I hope that continuous improvements can help you use deskpi pro smoothly. Thank you for your suggestions. We will try our best to do better.

Meusique, 2024-07-23:
Thanks for your work on this, @yoyojacky! I would definitely second making sure that all versions of the install script check that `dtoverlay=dwc2,dr_mode=host` is in the `[all]` section of `config.txt`, because as mentioned, it's under `[cm5]` by default on current versions of RaspberryPiOS. The current version of the script won't catch that it's in the wrong section.

I've switched to DietPi for now since it's a slicker fit for my use case than vanilla RaspberryPiOS, and the script works fine there because its default `config.txt` doesn't have `dtoverlay=dwc2,dr_mode=host` at all. But a version of the script that makes sure it's the right section couldn't hurt!

magorium, 2024-07-28:
@yoyojacky:
Sorry for the delay.

I have read your comments but unfortunately I could not understand the (which I assume are) Chinese parts.

I have everything running since my previous last post so that isn't my issue (anymore).

The issue I (currently) have is that the script itself is not working as advertised. I am personally in the position to rectify any mistakes but for ordinary users this might not be the case.

As far as I am able to tell the only issue left now (because you seem to have addressed the other issues, many thanks for that) is that the dtoverlay entry is checked against the wrong filtering rule and therefor will not be added on/to systems that run/install Raspi OS bookworm.

I'll leave my report open until that is fixed.

In case you require help fixing the script to fix that issue then please let us know so that we are able to assist (I have solved it myself by adding a new pi5 filter section and adding the dtoverlay entry there but using the all filter section should work as well)

magorium, 2024-08-31:
> Hi Bro, Is that you mean captrue the [all] section and then adding the dtoverlay line behind this section ?

That is the tricky part.

The answer to that question is yes for the standard bootconfig.txt that the raspberry provides with its bookworm distribution.

The answer is no for those who (manually) tinkered with the bootconfig.txt and/or uses another distribution (I have seen examples from some people in the issues that show a completely different bootconfig.txt

[](url)

As already mentioned the filtering rules are explained [here](url=https://www.raspberrypi.com/documentation/computers/config_txt.html#conditional-filters) and the [all] filtering section is able to create a ambiguous situation for handling the bootconfig.txt. And that is not even mentioning the [non] filter.

The current implementation of the script does not work. Not for Debian Bookworm at least and by seeing the issues mentioned for this repository also not for people using other distributions. 

Some of the opted solutions in this issue-thread are able to improve the situation (for Debian bookworm default bootconfig.txt) but also do not address all possible situations/combinations.

I am even unable to solve the ambiguity in real code, let alone that a bash script is able to solve that.

And all the above isn't a real issue, **as long as it is mentioned in the readme on the front page of this repository and communicated to the end-user by the script during installation**

Manually adding and/or moving the overlay entry is less than 5 seconds of work for most people but not if you as an end-user do not have a clue whatsoever that you need to do that (or at least verify if the overlay entry is present at the correct filtering section).

> but as my test, once you put the `dtoverlay=dwc2,dr_mode=host` to the config.txt file, it will take effect once your reboot Raspberry Pi, Is there some tricks on placing the paramater on config.txt file ?

No there actually isn't a trick. As explained above it needs to be added to the correct filtering section. That is why I personally opted to add  the [pi4] filtering section and added the dtoverlay entry there (but that is purely my own personal preference).
 
For "normal" Debian Bookworm users, adding the overlay to the [all] filtering section should be enough to make it work but, only for these people and those who did not manually tinkered with the bootconfig.txt contents.

So, I state again: As long as the end-user is not informed about this situation and the script does not correctly add the dtoverlay parameter to the correct filtering section, I'll have to leave this issue open.

See for example #186 to proof the point.

regards

yoyojacky, 2024-09-18:
In the boot process of the Raspberry Pi, the `/boot/firmware/config.txt` file is used to configure some parameters and hardware settings when the system starts. The `dtoverlay` parameter is used to dynamically load or configure specific hardware device's Device Tree overlays.

In the `config.txt` file, different sections are defined by square brackets `[]`, such as `[all]` or `[pi4]`. Each section can contain multiple parameters that are read and applied by the system when it starts.

1. **[all]**: Parameters in this section apply to all supported Raspberry Pi models. Regardless of the Raspberry Pi model, these parameters will be loaded as soon as the system starts.

2. **[pi4]**: Parameters in this section only apply to the Raspberry Pi 4 model. These parameters will only be loaded when the system detects that it is running on a Raspberry Pi 4.

Regarding the impact of placing the `dtoverlay` parameter under `[all]` and `[pi4]`:

- **Placing under [all]**: This means that regardless of the Raspberry Pi model, the `dtoverlay` parameter will be loaded as soon as the system starts. This could lead to loading unnecessary or incompatible hardware configurations on Raspberry Pi models that do not support it, potentially causing boot issues or system instability.

- **Placing under [pi4]**: This ensures that the `dtoverlay` parameter will only be loaded when the Raspberry Pi 4 model starts. Doing so can prevent loading incorrect hardware configurations on incompatible devices, thereby improving system stability and compatibility.

Therefore, if your `dtoverlay` parameter is specific to the hardware configuration of the Raspberry Pi 4, it is best to place it under the `[pi4]` section. This ensures that only the correct Raspberry Pi model will load these parameters, avoiding issues on incompatible devices.

In summary, whether to place the `dtoverlay` parameter under `[all]` or `[pi4]` depends on whether your hardware configuration is specific to the Raspberry Pi 4 or if it needs to be universal across all supported Raspberry Pi models. For model-specific configurations, it is recommended to use model-specific sections to enhance system stability and compatibility.
For the DeskPi Pro from versions v1 to v3, they only support the Raspberry Pi 4B, and other models simply cannot be used with this case. Therefore, from a functional perspective, there is no difference between placing the configuration under [all] and placing it under [pi4]. As long as it is placed at the end of your config.txt document, the functionality can be achieved. I'm not sure what you are concerned about?

regards

magorium, 2024-12-07:
Something with a brick wall :)

> The [all] filter
>
> The [all] filter is the most basic filter. **It resets all previously set filters and allows any settings listed below it to be applied to all hardware**. It is usually a good idea to add an [all] filter at the end of groups of filtered settings to avoid unintentionally combining filters (see below).

That indicates for every [all] filter that the configuration file contains and, which is per definition ambiquous (and yes, some distributions actually do that). And that is not even mentioning the fact that you can have several sections which are named similar. The official documentation does not mention what would or should happen in such cases (does it take the latest section as active one, does it collect all previous similar named sections, etc).

Unless that is made clear by the Raspberry Pi developers it is impossible to solve this (let alone with a simple bash script). Please check some of the other reports that clearly show issues with this approach.

Thus, as I propose for the umptieth time: mention it on the help/readme pages and be done with it instead of dragging this along and wasting both of our time.

regards

yoyojacky, 2025-02-06:
> Something with a brick wall :)
>
>
> That indicates for every [all] filter that the configuration file contains and, which is per definition ambiquous (and yes, some distributions actually do that). And [...]

just use the installation shell script, it has been updated for a while.

closed 2025-02-06 by yoyojacky