DeskPi-Team/deskpi #184: Minimum gLibC version required (2.34) is not available for Raspberry Pi OS

VoidJuiceConcentrate, 2024-07-27:
In trying to call pwmFanControl64 directly, I get the following error: 

`pwmFanControl64: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by pwmFanControl64)`

Yet, the highest available (stable) version of gLibC is 2.31, from official Debian sources. 

Either the tool needs to be built with gLibC 2.31 compatibility, or a workaround needs to be provided.

VoidJuiceConcentrate, 2024-07-28:
```pi@rpi4:~/ShinobiNodePrinter $ systemctl status deskpi.service
* deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/etc/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-07-27 23:10:29 EDT; 2min 58s ago
    Process: 1503560 ExecStart=/usr/bin/pwmFanControl64 & (code=exited, status=1/FAILURE)
   Main PID: 1503560 (code=exited, status=1/FAILURE)
        CPU: 3ms

Jul 27 23:10:29 rpi4 systemd[1]: Started DeskPi Fan Control Service.
Jul 27 23:10:29 rpi4 pwmFanControl64[1503560]: /usr/bin/pwmFanControl64: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /usr/bin/pwmFanControl64)
Jul 27 23:10:29 rpi4 systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
Jul 27 23:10:29 rpi4 systemd[1]: deskpi.service: Failed with result 'exit-code'. ```

Functionality is... spotty at the moment. Sometimes setting custom temperature and fan speeds works and sometimes it doesn't, this includes the default/number 7 option. Fan doesn't seem to respond to changing CPU temp outside of when initially set by the command deskpi-config.

magorium, 2024-07-28:
Hi @VoidJuiceConcentrate,
> Yet, the highest available (stable) version of gLibC is 2.31, from official Debian/RaspberryPi OS sources.

For your sake I hope it does not. Debian Bookworm itself is at 6.1.4 (and still counting up as we speak)

Did you by any chance cross compiled those tools or used another version of Raspi OS to compile them ?

VoidJuiceConcentrate, 2024-07-28:
No, this is a fresh install of Raspberry Pi OS. (plus updates before installing). 

```
pi@rpi4:~ $ uname -a
Linux rpi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@piholeultra:~ $ sudo apt install libc6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libc6 is already the newest version (2.31-13+rpt2+rpi1+deb11u10).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Not sure how to proceed here.

magorium, 2024-07-29:
Can you acknowledge that you are running Raspi OS bullseye ?

This is my setup:
```
$ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

$ ldd --version
ldd (Debian GLIBC 2.36-9+rpt2+deb12u7) 2.36
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

If you are running Bullseye then the utilities pwnControl64 and safeCutOffPower64 are required to be compiled manually see also https://github.com/DeskPi-Team/deskpi/blob/master/installation/drivers/README.md

Another possible solution is to update your current installation.

VoidJuiceConcentrate, 2024-07-30:
```
pi@rpi4:~ $ uname -a
Linux rpi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

pi@rpi4:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

pi@rpi4:~ $ ldd --version
ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u10) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

I did try to recompile them following the instruction inside, but the same error occurs on the newly built files.

magorium, 2024-07-30:
> I did try to recompile them following the instruction inside, but the same error occurs on the newly built files.

Nee impossible when you build the tools on/for a bullseye system.

After compilation of the tools and after downloading the installation repository the files in the driver/c directory need to be replaced with those that were manually build. Then the installation script install-raspios-64bit.sh can be run.

yoyojacky, 2025-07-21:
the pwmFanControl64 == 64bit executable binary file, it compiled in Raspberry Pi OS 64bit (bookworm) , you can just download the latest version of preinstalled image : https://drive.google.com/file/d/1OBY8g-3uRAPx2LAaGksvA-XAfwpPw7OU/view?usp=sharing

closed 2025-07-21 by yoyojacky