DeskPi-Team/deskpi #186: Missing /dev/ttyUSB0 on RaspberryOS with 6.6.31

walzing, 2024-08-21:
Hi,
it all worked in the past when the system was running on an earlier kernel.
Yesterday I reinstalled and did not find the CH340 Serial
```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And this what is expected.
[image]

This is the config.txt
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]
dtoverlay=dwc2,dr_mode=host
```
The config.txt after reinstall contained the dtoverlay unter [cm5]

This is the actual kernel:

`Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

yoyojacky, 2024-10-21:
CH340 is your serial port, you can use it to control your fan,  adding : `dtoverlay=dwc2,dr_mode=host` is to enable front USB port, you mean you want to use it to connect your serial USB device ?

yoyojacky, 2024-10-21:
Did you reboot your Raspberry Pi ?

yoyojacky, 2024-10-21:
execute :  `sudo ls /dev/ttyUSB* ` 
and paste the feedback to me, please.

Erpie, 2025-01-05:
Hi @yoyojacky , I have a similar problem.
Latest RaspberryOS (64bit) & updates (also from your side).
Fan control does not work.
I see the 2.0 root hub (bus 3), but no CH340 serial controller is detected. 
`lsub:
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
in /dev/ I don't see any ttyUSB-folder
`admin@raspi4bd:~ $ sudo ls /dev/ttyUSB0
ls: cannot access '/dev/ttyUSB0': No such file or directory`
Other info:
`admin@raspi4bd:~ $ dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:44:12:40 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=a795690a-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=BE
[    0.000329] printk: console [tty1] enabled
[    1.867956] fe201000.serial: ttyAMA1 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    1.868065] serial serial0: tty port ttyAMA1 registered
[    1.869593] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 37, base_baud = 62500000) is a 16550
[    3.865912] systemd[1]: Created slice system-getty.slice - Slice /system/getty.`
Also, my config.txt file is changed according to the instructions I read:
`[cm5]
#dtoverlay=dwc2,dr_mode=host

[all]
enable_uart=1
dtoverlay=dwc2,dr_mode=host
`
I tried with boot-SSD via USB-interface, as well as boot-SD-card (without any other USB-devices attached). Same problem in both situations. When I plug in another device (eg Logitec Unifier Receiver), it is only detected via the native PI usb-ports, NOT via the extra front-USB-connecters from Deskpi Pro:
`admin@raspi4bd:~ $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

yoyojacky, 2025-02-06:
could you please try to download this pre-installed image and flash it to your TF card and try again, I need to make sure that your hardware is working properly.  [Pre-Installed-image-download](https://drive.google.com/file/d/1fKJ6ZpMaEAjIvEa72R3Y0AAQbiwLF76c/view?usp=drive_link)

Erpie, 2025-03-04:
Thank you very much for your reply & sorry for my late reaction. 
OK, I tried the image and I see the QinHeng CH340 serial converter now (see dump below). Fan control is working, as well as the front USB-ports. Big leap forward. Thank you! 
My only problem: I cannot get it working with the SSD I intented to use (SP A55 / 1TB SATA). Each time I connect it with the SSD-PCB, screen gets blank and nothing happens (no white led neiher). When I use an older SSD-drive everything works and the system boots up, white led shows activity etc..  Remarkable: when I connenct the 1TB 'problem drive' with an external USB3->SATA cable, I works and I can boot from it. It seems the SSD-pcb cannot handle it (although I performed the firmware upgrade)..
Is there anything else I can check? Can I obtain the upgrade PCB evantually?
 
`pi@raspi4bd:~ $ lsusb
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 006: ID 13fd:3960 Initio Corporation INIC-3639
Bus 002 Device 009: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

yoyojacky, 2025-07-21:
> Thank you very much for your reply & sorry for my late reaction. OK, I tried the image and I see the QinHeng CH340 serial converter now (see dump below). Fan control is working, as well as the front [...]

Hello sir, 
I have build an preinstalled image for deskpi pro, you can download it from here: https://drive.google.com/file/d/1OBY8g-3uRAPx2LAaGksvA-XAfwpPw7OU/view?usp=sharing , and  as your description, I think the ssd may not compatible with ASM1053E SATA brige? could you please make sure the ssd is SATA protocol ? it the ssd is NVMe SSD , it will not be recognized by the daughter board of deskpi pro.