DeskPi-Team/deskpi #192: deskpi-cut-off-power.service hangs indefinitely during shutdown

akinnee, 2025-01-19:
On the Pi 4 B, with the original DeskPi Pro, fresh install of Raspberry Pi OS, correct /boot/firmware/config.txt, and freshly installed deskpi-config, shutdown never finishes. It hangs indefinitely waiting for `deskpi-cut-off-power.service` to exit.

How would I begin to troubleshoot this?

If I trigger the service manually, I don't get any logs, but it does shut the pi off after corrupting my shell history file.

bryanhrobbins, 2025-01-19:
I was having same problems after reinstalling raspberry pi os lite on my deskpi pro V2. I manually have to copy files to /user/bin and cut and paste the line in /boot/firmware/config.txt that was under [cm5] section to [all] section. Now when I type sudo poweroff it shutdown and not hang on the safe power off script.

akinnee, 2025-01-20:
The line that adds the deskpi overlays? Yeah, I had to manually add that as well. Front USB works. `deskpi-cut-off-power.service` works like half the time. It's weird.

akinnee, 2025-01-31:
After using it for a while, I'd say the safe shutdown service works like 90% of the time actually. I suppose a good bandaid is to add a timeout to that service's config to avoid a situation where I have to walk over and manually shut it off if it hangs.

yoyojacky, 2025-07-21:
it will be depraceted when the new script has been launched. you can ignore this service, it will removed in next image launched.

closed 2025-07-21 by yoyojacky