DeskPi-Team/deskpi #197: CH340 failed to connect to ttyUSB0 (Raspberry Pi OS Bookworm)

ExcitePrime, 2025-07-04:
Hello, can you help me solve this? The CH340 is already detected but it failed to connect to ttyUSB0

`
joshrpi4@raspberrypi:~ $ lsusb
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c534 Logitech, Inc. Nano Receiver
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
joshrpi4@raspberrypi:~ $ sudo dmesg |grep ttyUSB0
[    4.411247] ch341-uart ttyUSB0: probe with driver ch341-uart failed with error -32

`

ExcitePrime, 2025-07-04:
i also cannot find /dev/ttyUSB0

ExcitePrime, 2025-07-30:
> Try this preinstalled image : https://drive.google.com/file/d/1OBY8g-3uRAPx2LAaGksvA-XAfwpPw7OU/view?usp=drive_link

is it recommended to use the bookworm version of the raspberry pi os? i haven't tried the pre-installed image yet, and i am running on bullseye 64-bit, and i get this error when checking the status for deskpi.service

```shell
● deskpi.service - DeskPi Fan Control Service
     Loaded: loaded (/etc/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2025-07-30 16:03:32 PST; 21s ago
    Process: 670 ExecStart=/usr/bin/pwmFanControl64 & (code=exited, status=1/FAILURE)
   Main PID: 670 (code=exited, status=1/FAILURE)
        CPU: 2ms

Jul 30 16:03:32 raspberrypi systemd[1]: Started DeskPi Fan Control Service.
Jul 30 16:03:32 raspberrypi pwmFanControl64[670]: /usr/bin/pwmFanControl64: /lib/aarch64-linux-gnu/libc.so.6: versi>
Jul 30 16:03:32 raspberrypi systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
Jul 30 16:03:32 raspberrypi systemd[1]: deskpi.service: Failed with result 'exit-code'.

```

sfatula, 2025-08-07:
Is there a fix for an existing system, a patch, manual edit, whatever? I don't want to use that image and have to start over.

yoyojacky, 2025-08-19:
I have test it too, the driver of CH341 serial has some bugs . we are working on this , the raspberry pi OS 64bit (2025-05-13) release  can not drive the device well, we are trying to find a solution ASAP.