DeskPi-Team/deskpi #198: Power button state not changing to off on DietPi 64-bit after kernel update to 6.12.34+rpt-rpi-v8

986ster, 2025-07-04:
After updating a Pi 4B running DietPi 9.14.2 64-bit to kernel 6.12.34+rpt-rpi-v8, the power state LED is no longer recognizing when the Pi is powered off from the command line and remains in the blue state when the Pi is powered down.

This is when running the latest deskpi commit.

986ster, 2025-07-05:
Having dug further into this, I actually think this is a kernel bug because the /dev/ttyUSB0 device isn't being created when the firmware is updated, so I've opened a bug report with the Raspberry Pi OS team.

yoyojacky, 2025-07-21:
please try to download this pre-installd image and try again :https://drive.google.com/file/d/1OBY8g-3uRAPx2LAaGksvA-XAfwpPw7OU/view?usp=drive_link